ArduPilot/ardupilot #17245: Loiter Drifting to one side on FlywooF745

NitinJSanket, 2021-04-21:
## Bug report

**Issue details**
The quadrotor drifts to one side in Loiter but flies well in AltHold and holds position in FlowHold. 

**Version**
4.1.0dev

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad X4

**Hardware type**
Flywoo F745

**Logs**
Cannot log as it has no SD card support.

rmackay9, 2021-04-21:
Thanks for the report.  Really though these reports should go into the support forums, in particular the [Copter-4.1.0-beta1](https://discuss.ardupilot.org/c/arducopter/copter-41/159) testing thread although it looks like perhaps you're running "latest" instead of "beta1".  In general we recommend sticking with the betas because it is very hard for the dev team to provide general support for "latest".

I'll close this issue now, sorry to be strict but if we're not our to-do list (which is what this is) simply turns into another support forum.

By the way, it should be possible if there is a good telemetry connection to setup logging over the telemetry link.  Certainly we somehow need to get more detailed information in order to provide good feedback I think.

closed 2021-04-21 by rmackay9

rmackay9, 2021-04-21:
By the way @peterbarker tells me that the board should have a "blackbox" on it of 16MB so it should be possible to download onboard logs.

NitinJSanket, 2021-04-22:
> By the way @peterbarker tells me that the board should have a "blackbox" on it of 16MB so it should be possible to download onboard logs.

I did try it but it doesn't seem to work.

rmackay9, 2021-04-22:
@NitinJSanket,

We can investigate if the Flywoo board's onboard logging seems to be broken but if you've never downloaded an onboard log before we have [instructions here on the wiki](https://ardupilot.org/copter/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html).  Sorry if you've done this many times before and I'm telling you what you already know.

NitinJSanket, 2021-04-22:
> @NitinJSanket,
>
> We can investigate if the Flywoo board's onboard logging seems to be broken but if you've never downloaded an onboard log before we have [instructions here on the [...]

I've done it multiple times before, no logs are showing up though, even tried logging when disarmed, seems to not work.

lavachemist, 2021-04-22:
I assisted @NitinJSanket today, it was a param issue, not a firmware target issue.

LOG_BACKEND_TYPE was set to 1, when it should have been set to 4.

rmackay9, 2021-04-22:
nicely done @lavachemist!  Thanks!  I wonder if we need to change some parameter defaults... it seems unlikely that @NitinJSanket went in and changed this parameter himself..

lavachemist, 2021-04-22:
> nicely done @lavachemist! Thanks! I wonder if we need to change some parameter defaults... it seems unlikely that @NitinJSanket went in and changed this parameter himself..

I think that the parameter change was due to copying parameters from another FC, but I'm not sure. The default is correct and it works properly.