ArduPilot/ardupilot #17248: Lockup, EKF problems, and Internal Error from directory listing when max number of logs

pkocmoud, 2021-04-22:
## Bug report

**Issue details**

A Pixracer Pro user reported that "I reflashed firmware to stock & recalibrated everything just to make sure, but it appears to be a HW problem. We have at least 1, maybe 2 other autopilots that are starting to exhibit similar problems.

I've had this one just sitting on my desk powered for about 10 minutes & it currently thinks it's moving at about 700 m/s

So I figured out what it was. there was a fragmented log that was maxing out the number of logs & if you tried to grab the logs the autopilot would lock up. When it would recover the EKF would go nuts for a little while before it calmed down. Our companion computer was continuously trying to grab the logs so it never had a chance to recover."

They provided me a ZIP of their SDCARD and I was able to confirm the issue on Copter 4.0.7 and then on ArduCopter V4.1.0-dev (6235f7f8) 

While testing on master I also received the following internal error:

"PreArm: Internal errors 0x400 l:234 cnstring_nan"

I discussed with Peter Barker and he suggested I submit this bug report. Possibly relating to the Gaussian Sum Filter?

**Version**
Copter 4.0.7 and ArduCopter V4.1.0-dev (6235f7f8) 

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
NA

**Hardware type**
PixRacer Pro and Control Zero H7

**Logs**

This log contains the Internal error.
[00000007.BIN](https://drive.google.com/file/d/1y3iJW5Du3BJyrs_epp3svObEdykJneYU/view?usp=sharing)

rmackay9, 2021-04-22:
Yes, I think whenever a user downloads logs the main loop slows down and this makes the EKF happy.

almatthew, 2021-05-14:
This is an issue I've observed on an MRo Control Zero F7 as well. Though there is a clear workaround, its very disconcerting to see the issue appear, and makes the user think there is something wrong with the IMUs or the autopilot itself. I would propose a fix PR, but I'm not sure of the best approach for a fix.