ArduPilot/ardupilot #17257: Can't create folders through MAVFtp on Pixhawk mini

rizvanovic, 2021-04-22:
## Bug report

**Issue details**

_Trying to create a "scripting" folder in the APM directory results in an error message (see attached image). I could not find a way around this (I tried on different computers and versions of MP). Initially I feared the flash memory was too small on the PH mini but I haven't found any info confirming this.  _
[image]

**Version**
4.1.0 DEV Arduplane

**Platform**

[  ] All
[  ] AntennaTracker
[  ] Copter
[x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_Tailsitter_

**Hardware type**
_PIxhawk mini 3DR (bought from holybro)_

IamPete1, 2021-08-16:
related to https://github.com/ArduPilot/ardupilot/issues/14860

tridge, 2021-11-15:
fixed by #19267

closed 2021-11-15 by tridge