ArduPilot/ardupilot #17264: RTL goes crazy, very high risk of stall, i found the bug

StonedDawg, 2021-04-24:
when setting rtl alt to -1, (keep current altitude as RTL altitude), and also having a value under RTL_CLIMB, will make the plane climb up and cut throttle down to 0% to drop back to RTL altitude (altitude when RTL was triggered).

solution : dont set RTL_CLIMB when using RTL_ALT with a valur of -1. this should be safeguarded or implemented better to not risk wing to stall and crash

vierfuffzig, 2021-05-04:
this sounds like more of a support issue. 
@StonedDawg please post a log if you think you see any unexpected behaviour.

closed 2021-05-04 by vierfuffzig

StonedDawg, 2021-05-04:
@vierfuffzig I cant do the test again for a while, since i broke my usb port... will replace it soon, but just want to report and warn the others when setting rtl_alt to -1, do not set RTL_climb or plane will risk going into stall since it quickly dives down (after climbing to RTL_CLIMB) to last alt position before RTL is triggered.

vierfuffzig, 2021-05-04:
@StonedDawg this is the behaviour we see as expected when setting ALT_HOLD_RTL = -1 and any value for RTL_CLIMB:

[image]

plane will climb to the current altitude when RTL was triggered + RTL_CLIMB, then turn to home and descend on a calculated glidescope to reach the initial altitude when back home. it will NEVER dive down 90° in any automatic mode unless you've cranked up the limits (LIM_PITCH_MIN and _MAX and other speed-height-parameters) to allow it to do so.

i suggest you take this to the support forum https://discuss.ardupilot.org/ 
on a sidenote, adding a flight log is mandatory, no chance to tell what the issue is without one.

StonedDawg, 2021-05-04:
@vierfuffzig ah ok great thanks! i will go to discuss too... and i will get log too soon... thankyou for also running the simulator to test it!