ArduPilot/ardupilot #17266: Smartaudio VTX control can't change band/channel

griffondrones, 2021-04-24:
## Bug report

**Issue details**

When using smartaudio VTX control with a TBS UNIFY PRO32 HV VTX, changing bands or channels doesn't seem to work.
The connections and configuration appears okay, since pitmode works. 

**Version**
Latest 4.1.0 dev-version available 2021-04-24. ArduPlane build.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Flying wing

**Hardware type**
Matek F405-Wing
TBS UNIFY PRO32 HV

**Logs**
Nothing in the log points to a problem.

andyp1per, 2021-04-24:
Please can you post your parameters

griffondrones, 2021-04-24:
Ah, never mind, it seems like it sorted itself out somehow.  Possibly a bad connection for the SA pin..

closed 2021-04-24 by griffondrones