ArduPilot/ardupilot #17269: Serial passthrough does not work for serial ports configured to "GPS" with GPS protocol set to ublox

Sebbb, 2021-04-25:
## Bug report

**Issue details**

Serial passthrough does not work for serial ports configured to "GPS" with GPS protocol set to ublox

With the settings
SERIAL3_PROTOCOL = 5
GPS_TYPE=2
SERIAL_PASS1=8
SERIAL_PASS2=3
I only receive a few packets every 10 seconds

After setting 
SERIAL3_PROTOCOL = -1
and rebooting it's working fine.

**Version**
4.1-dev

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

(tested only with Copter, probably affecting other platforms as well)

**Airframe type**
irrelevant

**Hardware type**
Matek H743

**Logs**
output of u-center packet console, where you should see packets all the time, however, there are big gaps between the time stamps

19:56:49  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:56:49  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:56:49  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:56:49  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:56:49  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:56:49  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:56:49  R -> UNKNOWN ???,  Size 359,  'UNKNOWN'
19:56:49  R <- UBX MON,  Size   8,  'Monitor'
19:56:58  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:56:58  R <- UBX MON,  Size   8,  'Monitor'
19:56:58  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:56:58  R -> UBX MON-VER,  Size 228,  'Version'
19:56:58  R -> UBX MON-VER,  Size 228,  'Version'
19:56:58  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:56:58  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:56:58  R -> UBX MON-HW,  Size  68,  'Hardware Status'
19:56:58  R -> UBX MON-HW2,  Size  36,  'Extended Hardware Status'
19:56:58  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:56:58  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:56:58  R -> UBX NAV-TIMEGPS,  Size  24,  'GPS System Time'
19:56:58  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:56:58  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:56:59  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:56:59  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:56:59  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:56:59  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:56:59  R -> UNKNOWN ???,  Size 377,  'UNKNOWN'
19:56:59  R <- UBX MON,  Size   8,  'Monitor'
19:57:07  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:07  R <- UBX MON,  Size   8,  'Monitor'
19:57:07  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:07  R -> UBX MON-VER,  Size 228,  'Version'
19:57:07  R -> UBX MON-VER,  Size 228,  'Version'
19:57:08  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:08  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:08  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:08  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:08  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:08  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:08  R -> UBX MON-HW,  Size  68,  'Hardware Status'
19:57:08  R -> UBX MON-HW2,  Size  36,  'Extended Hardware Status'
19:57:08  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:08  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:08  R -> UBX NAV-TIMEGPS,  Size  24,  'GPS System Time'
19:57:08  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:08  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:08  R -> UNKNOWN ???,  Size 481,  'UNKNOWN'
19:57:08  R <- UBX MON,  Size   8,  'Monitor'
19:57:17  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:17  R <- UBX MON,  Size   8,  'Monitor'
19:57:17  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:17  R -> UBX NAV-TIMEGPS,  Size  24,  'GPS System Time'
19:57:17  R -> UBX MON-VER,  Size 228,  'Version'
19:57:17  R -> UBX MON-VER,  Size 228,  'Version'
19:57:17  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:17  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:18  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:18  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:18  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:18  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:18  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:18  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:18  R -> UBX MON-HW,  Size  68,  'Hardware Status'
19:57:18  R -> UBX MON-HW2,  Size  36,  'Extended Hardware Status'
19:57:18  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:18  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:18  R -> UBX NAV-TIMEGPS,  Size  24,  'GPS System Time'
19:57:18  R -> UNKNOWN ???,  Size 352,  'UNKNOWN'
19:57:18  R <- UBX MON,  Size   8,  'Monitor'
19:57:27  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:27  R <- UBX MON,  Size   8,  'Monitor'
19:57:27  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:27  R -> UBX MON-VER,  Size 228,  'Version'
19:57:27  R -> UBX MON-VER,  Size 228,  'Version'
19:57:27  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:27  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:27  R -> UBX NAV-TIMEGPS,  Size  24,  'GPS System Time'
19:57:27  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:27  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:28  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:28  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:28  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:28  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
19:57:28  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
19:57:28  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'

(I know, there is this "MAVSerial Pass" feature in Mission planner, however, serial passthrough should also work)

rmackay9, 2021-04-26:
We really need to get support questions raised in the forums before being raised as issues.  Any chance you could create a new topic in the appropriate 4.1 category depending upon the vehicle?  Copter-4.1 is here: https://discuss.ardupilot.org/c/arducopter/copter-41/159

rmackay9, 2021-05-28:
I've reproduced this issue with both Copter-4.0 and 4.1.0-beta3 using [serial pass through as described here on the wiki](https://ardupilot.org/copter/docs/common-serial-passthrough.html).

As mentioned above we can see that the packets from the GPS are very intermittant.  We get packets for a few seconds then they stop arriving for perhaps 10 seconds, then they restart again.

[Here is a short video showing what we are seeing](https://youtu.be/kedv7tlysHI)

rmackay9, 2021-06-01:
Hi @Sebbb,

We discussed this on the dev call and figured out the issue.  The underlying problem is that ArduPilot's GPS driver is still running and so it is sometimes consuming the GPS output.  So this means in u-center you sometimes get data but other times that data goes to the GPS driver.

The solution is to disable the GPS driver's use of the serial port by changing SERIAL3_PROTOCOL  = -1 (None).  I also found that I needed to set SERIAL3_BAUD = 230400 and I knew this was the correct baud rate by checking the results of the GPS driver's "auto bauding" that it does soon after startup.
[image]

I've added these instructions to the [Serial Passthrough Wiki page](https://ardupilot.org/copter/docs/common-serial-passthrough.html) (changes should appear within 30min of this post).  Maybe you could try and see if this resolves the issue you're seeing?  If not maybe I can make the wiki clearer.

Sebbb, 2021-06-01:
Hi Randy,

yes, I figured out that SERIAL3_PROTOCOL = -1 will solve it, as written in the original problem description. As I migrated to an UAVCAN GPS, this isn't relevant for me anymore, but it took me some hours to find out what's wrong, and that it is not a hardware issue. So it's good, that this is documented now :)

Is it planned to fix it, that it will also work without changing the PROTOCOL? SERIAL_PASS2 reverts automatically to -1, for the protocol I would need to do it manually.. 

Thx,
Sebastian

rmackay9, 2021-06-01:
@Sebbb,

At the dev meeting we did discussed two possibilities:

1. a lock on the serial port so that the pass through can block the driver from reading the packets
2. sending the incoming packets to both the serial driver and the pass-through

We didn't commit to making these changes though and I'm not sure if/when we will get to it.  It's not that they wouldn't be useful, it certainly would be, it's just about developer time and priority and with so much else on the to-do list I can imagine it will be a long time before we would get to this.

Txs for the feedback and the report, I'm going to close this now that the wiki is updated.

closed 2021-06-01 by rmackay9

Sebbb, 2021-06-01:
BTW, there's a small typo in the wiki page:

"For example, if attempting passthrough to the first GPS, SERIAL3_BAUD should normally be set to 230400 but you may check the GPS driver’s “autobauding” by looking at the ground stations Message window soon after **starup**"

rmackay9, 2021-06-02:
@Sebbb, great thanks for the feedback, typo fixed!

ROSStargh, 2023-06-01:
Hi @rmackay9 , 
I found that changing SERIAL1_PROTOCOL to -1 does not work. I had to set it to 2 (Mavlink2) to get it to work. Here is my [post](https://discuss.ardupilot.org/t/serial-passthrough-how-to-set-it-up/101768).

rmackay9, 2023-06-01:
[Serial passthrough is certainly broken in 4.3](https://github.com/ArduPilot/ardupilot/issues/21114) but [we have a fix](https://github.com/ArduPilot/ardupilot/pull/23888) that will certainly go into 4.4.0.  It is unclear if/when we will backport this fix to 4.3.