ArduPilot/ardupilot #17278: Message appears to be too long: AutoTune: failing to level, manual tune may be req....uired

lthall, 2021-04-27:
This message in Autotune appears to be too long making it hard to read:

27/04/2021 11:09:49 AM : uired
27/04/2021 11:09:49 AM : AutoTune: failing to level, manual tune may be req

[image]

WickedShell, 2021-04-27:
With MAVLink 2 and a reasonably up to date GCS you should see these combined into a single message, but it's definitely not always going to be combined, and on a MAVLink 1 system, it will always look as you outlined. It's your guys call as the autotune maintainer how much you care about it being incorrect on MAVLink 1.

peterbarker, 2021-04-28:
Well... the message chunks should really still be displayed in order IMO.

Pretty sure we spit them out in the right order :-)

lthall, 2021-04-28:
I assume there is a maximum number of characters that we should use before this happens. It looks like about 50. Is there a specific number we should try to stay below or will it be variable based on what is in front or something?

peterbarker, 2021-04-28:
50 is the magic number

But for low-bandwidth errors, clarity is a good thing for reducing user support.

Maybe if we poke @meee1 he might start reassembling messages.  Or emitting them in the right order :-)

rmackay9, 2021-04-28:
Let's just shorten it.  How about, "AutoTune: failing to leve, please tune manually"?

lthall, 2021-04-28:
Would it be acceptable to do a double message so it comes out as:
AutoTune: failing to level, 
manual tune may be required.

peterbarker, 2021-04-29:
> Would it be acceptable to do a double message

Ewww :-)

We have facilities to chunk/reassemble longer messages which should be used for doing that.

Additionally two messages may still suffer from that reordering problem in MissionPlanner.

@rmackay9 's plan is probably the best.

lthall, 2021-04-29:
> @rmackay9 's plan is probably the best.

I agree. 50 characters or less :)

PR #17295 (2021-04-29): autotest: shorten statustext message to avoid chunking
Closes #17278
closed 2021-04-29 by rmackay9