ArduPilot/ardupilot #17281: can't connect to /dev/ttyACM0 on ArchLinux (5.11.16)

booo, 2021-04-27:
## Bug report

**Issue details**

I connected ArduPilot/PixRacer to my x220 with latest ArchLinux and 5.11.16 Linux running. As soon as I try to open /dev/ttyACM0 with minicom or QGroundControl the kernel spits out an error message and I can't open the device:

```
cdc_acm 1-1.2:1.0: acm_port_activate - usb_submit_urb(ctrl irq) failed
```

**Version**
4.0.9
4.0.7

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

Talon, Fix Wing Airframe

**Hardware type**
PixRacer R15

**Logs**

Kernel log:
```
[  259.829740] usb 2-1.2: USB disconnect, device number 8
[  261.846050] usb 2-1.2: new full-speed USB device number 9 using ehci-pci
[  261.945500] usb 2-1.2: New USB device found, idVendor=26ac, idProduct=0012, bcdDevice= 1.01
[  261.945519] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  261.945526] usb 2-1.2: Product: PX4 BL FMU v4.x
[  261.945530] usb 2-1.2: Manufacturer: 3D Robotics
[  261.945535] usb 2-1.2: SerialNumber: 0
[  261.946221] cdc_acm 2-1.2:1.0: ttyACM0: USB ACM device
[  266.846092] usb 2-1.2: reset full-speed USB device number 9 using ehci-pci
[  266.942949] usb 2-1.2: device firmware changed
[  266.943258] usb 2-1.2: USB disconnect, device number 9
[  267.022742] usb 2-1.2: new full-speed USB device number 10 using ehci-pci
[  267.121477] usb 2-1.2: New USB device found, idVendor=1209, idProduct=5741, bcdDevice= 2.00
[  267.121494] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  267.121501] usb 2-1.2: Product: Pixracer
[  267.121506] usb 2-1.2: Manufacturer: ArduPilot
[  267.121511] usb 2-1.2: SerialNumber: 29003A001751353330343737
[  267.122219] cdc_acm 2-1.2:1.0: ttyACM0: USB ACM device
[  280.624683] audit: type=1100 audit(1619527799.176:99): pid=2861 uid=1000 auid=1000 ses=1 msg='op=PAM:authentication grantors=pam_faillock,pam_permit,pam_faillock acct="borgers" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  280.626794] audit: type=1101 audit(1619527799.179:100): pid=2861 uid=1000 auid=1000 ses=1 msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="borgers" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  280.627938] audit: type=1110 audit(1619527799.179:101): pid=2861 uid=1000 auid=1000 ses=1 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  280.630978] audit: type=1105 audit(1619527799.183:102): pid=2861 uid=1000 auid=1000 ses=1 msg='op=PAM:session_open grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  280.779893] cdc_acm 2-1.2:1.0: acm_port_activate - usb_submit_urb(ctrl irq) failed
[  280.780502] audit: type=1106 audit(1619527799.333:103): pid=2861 uid=1000 auid=1000 ses=1 msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  280.780586] audit: type=1104 audit(1619527799.333:104): pid=2861 uid=1000 auid=1000 ses=1 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
```

khancyr, 2021-04-27:
hello,

Check that you got enough power if your board is power some sensor.
This is a linux error, I will close this. You could try on https://discuss.ardupilot.org/ if somebody got a solution but you will have better luck at looking on the net or Arch forum.
Thanks

closed 2021-04-27 by khancyr

booo, 2021-04-28:
Thanks a lot for your feedback.

This happens even with a power source connected to the board.