ArduPilot/ardupilot #17286: Wrong calculation of distances in OBSTACLE_DISTANCE function

mtbsteve, 2021-04-28:
## Bug report
I am sending distances from the point cloud of a ZED stereo cam using the OBSTACLE_DISTANCE function to AP. The proximities however are calculated wrongly in AP. 
The ZED has a Horizontal-FOV of 86 degrees. I am generating 72 distance values from the middle of the view with a resulting increment_f of 1.194.
Now in the proximity viewer in MP, I see only values for the forward left and forward distance, but not for the forward right distance. While the forward right distances are missing, the left and forward distance are also wrong.

However, when I do not load the copter parameters in MP and open the proximity viewer, I see the 72 distance vectors correctly displayed with the correct distances over the entire camera H-FOV from -43 - +43 degrees. 
Since I can see the same wrong calculated distances in other GCS like Solex, I assume its not a problem in MP.

Note: sending just a single distance with the DISTANCE_SENSOR function works fine with the correct distance in the image center.
Screenshots:
Mavlink inspector with the correct data as sent from the TX2:
[https://1drv.ms/u/s!AnKeW8KMoCcygP8IB4HV6tRT_krxMQ?e=ludtc3](url)
Wrong display of proximity in the Proximity viewer:
[https://1drv.ms/u/s!AnKeW8KMoCcygP8HsrjkUKRnRGrSIg?e=H4dWQj](url)
And here is the correct representation of the 72 individual values in the proximity viewer with the copter parameters not loaded.
[https://1drv.ms/u/s!AnKeW8KMoCcygP8M7fLiM6WGyP8HqQ?e=b3a8Bu](url)
Last but not least the real image from the ZED cam in Solex: here the forward right distance is also wrong.
[https://photos.app.goo.gl/qB4aozweJZzmkRAH8](url)

**Version**
Copter 4.0.7, ZED camera, Jetson TX2 companion computer

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad
**Hardware type**
Cube black
**Logs**
https://drive.google.com/file/d/1KEhMS31LmprBs6b6Kv64yRd_gfV9ieGb/view?usp=sharing

rmackay9, 2021-04-28:
It might be a good idea to test with Copter-4.1.0-beta1 to see if that makes any difference.  Although I suspect it won't any fixes would be more likely to go out with 4.1 rather than a patch to 4.0.

mtbsteve, 2021-04-29:
Thanks @rmackay9 . I will retest in AC 4.1. Btw I just found the same issue reported by you and @thien94 by August last year in the MP issue list along with the Realsense cam: https://github.com/ArduPilot/MissionPlanner/issues/2444
I don't believe that its an issue in MP but rather a problem in Arducopter, since I am getting the same issue in other GCS, too. No need to patch 4.0.7, a fix in 4.1 would be great.

rmackay9, 2021-04-30:
@mtbsteve, great, thanks.  My understanding after discussing with @rishabsingh3003 is that it's fixed in 4.1.

IamPete1, 2021-08-16:
sound like this is fixed.

closed 2021-08-16 by IamPete1