ArduPilot/ardupilot #17291: SLCAN Mode CAN2 not enabling on ProfiCNC/Mini Carrier with Here3

jkreienh, 2021-04-28:
## Bug report

**Issue details**

I'm not able to update Here3 parameters or firmware via SLCAN interface using CAN2 on the ProfiCNC/Mini Carrier.  I see both Mavlink and SLCAN ports and have experimented with settings for SLCAN and I'm not able to view parameters or see anything happening with SLCAN after selecting SLCan Mode CAN2.  GPS works fine and fixes with RTK via NTRIP RTK/GPS inject.  I was able to update the Here3 firmware via SLCAN in 4.07 and wanted to confirm the firmware on the uBlox chip via passthrough in that version.  The TCP passthrough didn't work in 4.07 to uCenter, so I was attempting to try this again in 4.1 Dev after updating to enable LUA scripting.

[image]

**Version**
Copter 4.1 Dev

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
TBS Discovery Quad 'V'

**Hardware type**
ProfiCNC/HEX The Cube Orange
ProfiCNC/Mini Carrier (CAN2)
GPS1: Here 3+

jkreienh, 2021-05-02:
I just checked with v4.07 Stable and SLCan Mode CAN2 enables fine, so we must have a bug in v4.10 Dev

IamPete1, 2021-08-16:
This is possible, but a bit tricky,  you must be using driver 2 for port 2.