ArduPilot/ardupilot #17296: AP_periph: Node does not play tunes if flight controller has no alarm defined.

ukmook, 2021-04-29:
## Bug report

**Issue details**

If a flight controller without an alarm circuit is connect by CAN to a node with alarm circuit then then the node does not play any tunes from the sounder. 

Tests
1) The node has a pin defined in hwdef as PA1 TIM2_CH2 TIM2 GPIO(77) LOW ALARM The flight controller (autopilot) has no buzzer circuit so no alarm defined in hwdef. After testing no sound comes from the node. 

2) A pin was defined on the Flight controller PD12 TIM4_CH1 TIM4 GPIO(80) ALARM which is not connected to anything (no circuit exists). The node plays tunes from the sounder. 

So if a alarm is not defined in the hwdef of the Autopilot connected to the node then you get no sounds from the node.

Tested with: 
define HAL_PERIPH_ENABLE_NOTIFY
define HAL_PERIPH_ENABLE_RC_OUT

**Version**
4.1.0 Beta, Flight controller / Dev build on Node

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
n/a

**Hardware type**
Custom flight controller and custom Node

**Logs**
n/a

magicrub, 2021-04-29:
To clarify, this has nothing to do with the periph node. The problem is if there's no hardware alarm defined in the hwdef of the flight controller then the buzzer feature is disabled and ap_notify won't generate the UAVCAN packet.

bugobliterator, 2021-05-26:
@ukmook can you test the above PR, and see if it solves your issue.

ukmook, 2021-05-30:
Tested and working .. Thanks

closed 2021-06-01 by ukmook