ArduPilot/ardupilot #17314: EKF3 yaw timeout for GPS yaw

tridge, 2021-05-01:
When using GPS yaw we can get long periods where we have no yaw available, then yaw comes back. This particularly happens for rovers which might be sitting still for a while without good sky coverage. When GPS yaw recovers the EKF3 yaw may have drifted enough due to gyro drift to cause high innovations. The EKF3 should timeout and reset yaw with the new data when GPS yaw recovers.
An example replay log is here:
http://uav.tridgell.net/tmp/gps-yaw-recovery-ek3.bin
[image]

priseborough, 2021-05-04:
@tridge The log shared with this issue has a major problem with the first baro, eg:

[image]

[image]

This causes the EKF to learn an incorrect vertical accel bias and the associated innovations are also affecting the yaw gyro bias estimation.

andyp1per, 2021-05-04:
FWIW I am seeing altitude drift problems that might or might not be related to baro issues. On one copter with baro copiously covered in foam it will decide to suddenly descend and then at other times is perfectly fine. With another copter landed the vertical position innovations would grow rapidly and then get reset over and over. I don't fly much in non-manual modes so it doesn't affect me much, and almost certainly more to do with my builds but just thought I would mention it in case there is a deeper problem.

priseborough, 2021-05-04:
@andyp1per the first baro in this log is external to this autopilot - don't know anything beyond that. The two internal baro's were OK. I was able to work around the baro error during replay by using EK3_AFFINITY = 2 and ignoring replay results for the first EKF core.

jmachuca77, 2021-05-05:
Actually, I don't think I can disable it, all I seem to be able to do is set another baro as primary, There is no "BARO_USE" params like for compass.