ArduPilot/ardupilot #17332: Ardupilot 4.1 Beta FPort telemetry issue

Dronotique, 2021-05-02:
## Bug report

**Issue details**

I give a try to the 4.1.0 Beta, coming from 4.0.7 and I get telemetry issue with Frsky F.Port
My setup is a Kakute F7 with R9 MX OTA Receiver.
Radio side there is a FrSky Taranis XLite Pro and a R9M Lite Pro RF module.

All FrSky elements have last firmwares installed.

With Arducopter 4.0.7 all works fine.
With 4.1.0 Beta a got a lot of error "Lost Telemetry" and then the telemetry is recovered, and so .

You can see this on this short video :
https://youtu.be/kiTocs65BFE

**Version**
Arducopter V4.1.0 BETA

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Kakute F7 Mini

**Logs**
No logs, see video

khancyr, 2021-05-02:
Hello

Without a dataflash log we won't be able to do much. Wouldn't it be possible to provide one ?

Dronotique, 2021-05-02:
I Can tomorrow. But the error seems to be visible only on the transmitter

rmackay9, 2021-05-03:
It would be best to keep the 4.1 reports in the Copter-4.1 discussion so I've created one here: https://discuss.ardupilot.org/t/frsky-f-port-telemetry-issue-on-kakute-f7-with-r9-mx-ota-receiver/70747

I'll close this issue if that's ok.  Sorry to be strict but if we're not then the to-do list turns into a support forum.

Txs very much for testing 4.1!

closed 2021-05-03 by rmackay9

yaapu, 2021-06-16:
@Dronotique which version of OpenTX are you using and which RF (access, D16 v1, D16 v2) firmware on your TX/RX?

Dronotique, 2021-06-16:
> @Dronotique which version of OpenTX are you using and which RF (access, D16 v1, D16 v2) firmware on your TX/RX?

I'm using OpenTX 2.3.11 with ACCESS firmware