ArduPilot/ardupilot #17365: Quadplane, Plane 4.0.9: Flyaway on autonomous mode after letting the vehicle disarm in auto mode.

ohitstarik, 2021-05-05:
Hey. 

I'm running identical missions multiple times on a test vehicle. If I manually disarm the vehicle at the end of a mission (switch), i can bring the aircraft back to the takeoff spot and run the mission again no problems. Here's the vehicle running it 3 times back to back to back with no issues.

[image]

The moment i let the vehicle disarm itself, the next time I run the mission, the vehicle seems to run the first 2 waypoints just fine (VTOL Takeoff + do vtol transition to multicopter), but on the third (loiter at position), it will just completely ignore the waypoint and fly in a random direction. I havent been able to find a pattern to this. Here's the flight right after those two:

[image]

Usually i'd have to do a hard reboot and reload the mission. Here's another example:

[image]

Here's a log file. https://drive.google.com/file/d/1ksxiqw7DNqc4n2el0nIUKQO4OUyoBqDH/view?usp=sharing

Flight #1 is a different error that i've come across.

Flights #2, 3, and 4 are successful flights. I let the vehicle autonomously disarm after flight 4. The vehicle attempts to "fly away" (i have no idea where it's going) on flight 5. Flight 6 i let it autonomously disarm, and flight 7 it starts going backwards.

peterbarker, 2021-05-07:
Reasonably easy to find fault here, I think.

This is jut running the mission back-to-back with a `wp set` in the middle:

[image]

peterbarker, 2021-05-07:
Recreates on master but the logging is different (missing the spike in TPX)

[image]

peterbarker, 2021-05-07:
... sadly, I think that issue i spotted there is actually different to the OP's....