ArduPilot/ardupilot #17381: 3 or 2 point fine calibration of the current sensor

robustini, 2021-05-07:
It is well known that the current sensor in the Power Modules, even if calibrated, then detects incorrect measurements when exiting the calibration point.
It would therefore be interesting to have more calibration points, to ensure that in the intermediate steps the measured current is closer to the real one.
This method has always existed also in the temperature meters of 3d printers, where each thermistor has its own table with many preset measurements.
In Marlin and Klipper, but also in other firmware, the reading error can be corrected by acting on three or more points.
So it would be enough to do it also with own current sensors.
Practical example: if I know that my system works in a 10-60A current range, I perform three points calibrations, one at 10, the other at 35 and the last at 60A.
So imho the code with this steps should return a better read.

albertocopter, 2021-05-07:
Yes ...good and useful idea!

ik5jaumax, 2021-05-08:
Given the countless types of power modules on the market in which each of them returns a different result from the other, it would be a step forward in giving a certain accuracy of values. I therefore believe that an implementation of such a function would be well desirable

IamPete1, 2021-08-15:
closing in favor of https://github.com/ArduPilot/ardupilot/issues/6213 as it has more comments.

closed 2021-08-15 by IamPete1