ArduPilot/ardupilot #17383: ekf spamming forced reset warnings

jaxxzer, 2021-05-07:
[image]

This happens on ArduSub 4.x, when the autopilot is sitting still and disarmed. When the autopilot is moved or armed, the messages stop. The situation seems benign and everything else is functional, but the warning is annoying and concerning to users and implies that there is a problem with the hardware.

rmackay9, 2021-05-07:
@jaxxzer we will probably need to see an onboard log with LOG_DISARMED = 1.  FYI @priseborough

andyp1per, 2021-05-08:
@rmackay9 pretty sure this is the code that allows compass-less copters to get off the ground. We need the code otherwise the EKF gets stuck in initializing. One option would be simply to suppress the message when disarmed - not sure it adds any value.

jaxxzer, 2021-06-17:
Here is a log demonstrating the problem, the autopilot is just sitting on my desk during this test. https://file.io/JgT4sxIQOMMB

tridge, 2021-07-04:
there is a good chance this issue is already fixed in master by this PR: https://github.com/ArduPilot/ardupilot/pull/17787
if it is not fixed we need a replay log showing the issue to fix it. @jaxxzer your log link has expired.
A replay log is one with LOG_DISARMED=1 and LOG_REPLAY=1

spiderkeys, 2021-07-05:
@tridge Attached is a log that shows the issue on sub with LOG_DISARMED=1 and LOG_REPLAY=1, and was captured from a version of master more recent (https://github.com/ArduPilot/ardupilot/commit/819c331acdf4675ac5b6e2b519f1e22f8ed511be) than when PR: #17787 was merged.

Platform: MatekH743-Wing

[ekf3_resets.log](https://github.com/ArduPilot/ardupilot/files/6765212/ekf3_resets.log)

tridge, 2021-07-08:
> Attached is a log that shows the issue on sub with LOG_DISARMED=1 and LOG_REPLAY=1, and was captured from a version of master more recent ([819c331](https://github.com/ArduPilot/ardupilot/commit/819c331acdf4675ac5b6e2b519f1e22f8ed511be)) than when PR: #17787 was merged.

this log has INS_GYR_CAL=0 so no gyro cal on startup, and the first gyro is a fair way off zero. Can you see if calibrating the gyro fixes the issue?
Note that if you need INS_GYR_CAL=0 then it can be worthwhile doing a temperature cal of the IMUs so they compensate for drift due to temperature changes. See https://ardupilot.org/copter/docs/common-imutempcal.html

spiderkeys, 2021-07-08:
Hi @tridge, if the delta from zero you are seeing is in mainly in the pitch axis, with some in the roll axis, that is because I had my FMU pitched up. I do this such that I can easily access debug ports and other I/O while doing development. See below:

[image]

I thought at the time that the pitch might also be affecting the EKF behaviour, but I saw the same results with the vehicle perfectly level after running a gyro calibration.

INS_GYR_CAL being disabled is generally required with Sub, since the deployment and startup conditions can vary dramatically from what a copter might experience. Gyro cals are usually performed manually in a controlled environment before moving the vehicle elsewhere to begin the dive.

That said, it is good to know about the temperature calibration process. If you think this could be an important factor, I can perform that and run another log.

@Williangalvani might also have some relevant input here, as he was seeing a constant Yaw drift on EKF3 as well, despite magnetometer output not showing any significant movement. I believe I saw a similar issue in the Copter Todo list that I thought might be contributing as well:

https://github.com/ArduPilot/ardupilot/issues/17030

tridge, 2021-07-08:
> if the delta from zero you are seeing is in mainly in the pitch axis, with some in the roll axis, that is because I had my FMU pitched up

that affects accels, not gyro

> I saw the same results with the vehicle perfectly level after running a gyro calibration.

can you post a log reproducing this problem with INS_GYR_CAL=1 ?