ArduPilot/ardupilot #17388: Plane: Pitch to throttle reverse option in loiter mode

snowblind64, 2021-05-08:
## Feature request

**Is your feature request related to a problem? Please describe.**
In loiter mode, the pitch to throttle ratio is reversed from what would be expected when using stick mixing to influence the flight path. Pulling back on the elevator to pitch up and gain altitude reduces throttle. If held in this position, the aircraft will go into a stall.

**Describe the solution you'd like**
I believe an option to reverse the throttle to pitch ratio in loiter mode so that throttle increases as the aircraft pitches up, and decreases as it pitches down, would be a safer and more intuitive way to manage temporary altitude changes in loiter mode. 

**Describe alternatives you've considered**
If throttle to pitch cannot be changed for loiter mode, there should at least be a warning in the documentation for stick mixing regarding the potential to stall the airplane. Perhaps even have stick mixing turned off by default.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

magicrub, 2021-05-09:
Are you using stick mixing 1 or 2?

snowblind64, 2021-05-09:
I am using stick mixing 1.

magicrub, 2021-05-09:
Do you have a log you can share?

snowblind64, 2021-05-09:
I'd have to sort quite a few logs. I'm going out to fly this afternoon and will replicate the issue and upload a log.

snowblind64, 2021-05-10:
I've spent some time researching this and I believe I have an understanding as to the reason for the unwanted behavior. When stick mixing is used, TECS is not aware of this and sees the commanded altitude change as error. While stick inputs are active, TECS can only control throttle, and as such attempts to use it to return to it's altitude setpoint by increasing or decreasing throttle.

Here is a log from yesterday. I replicate the issue in the last couple minutes of flight while loitering in RTH (same issue happens in RTH loiter and auto modes iirc):
https://drive.google.com/file/d/1jCM2cKzohqPibgKm48XfR5IW48mmimi7/view?usp=sharing

IamPete1, 2022-05-22:
Loiter can now be controlled like FBWB using the `FLIGHT_OPTIONS` bit `12: Enable FBWB style loiter altitude control`

closed 2022-05-22 by IamPete1