ArduPilot/ardupilot #17401: Copter: "PreArm: Bad GPS Position" when using optical flow

rmackay9, 2021-05-10:
At least in SITL (and perhaps on a real vehicle) a "Bad GPS Position" pre-arm warning appears if using only optical flow.

The setup for optical flow needs to be updated on the [optical flow wiki page](https://ardupilot.org/copter/docs/common-optical-flow-sensor-setup.html) but the param setup is something like this:

- EK3_SRC1_POSXY = 0 (none)
- EK3_SRC1_VELXY = 5 (optical flow)
- EK3_SRC1_VELZ = 0 (none)

arBalasquide, 2021-07-02:
@rmackay9 Would checking that `EK3_SRC1_VELXY` is set to 5 and then returning true in `AP_Arming:gps_checks` be a valid fix?

Or do we want to confirm that indeed a GPS isn't connected and only then do we skip gps prearm checks?

tajisoft, 2022-02-01:
I faced same issue using T265. I got PreArm' GPS 1: Bad fix message on HUD.

EK3_SRC2_POSXY 6
EK3_SRC2_POSZ 1
EK3_SRC2_VELXY 6
EK3_SRC2_VELZ 6
EK3_SRC2_YAW 6

wildangunawan, 2022-02-17:
I see that in issue list that this issue is "nice to have". With current workaround, is there anything to watch for since we don't do any check before arming?

Or for safety reason, we can arm check = 1 first. Then see if only GPS warning, if yes then all good we can set arm check to 0 again.

rmackay9, 2022-02-17:
@wildangunawan,

I think for now the best solution is to enable all the arming checks except for GPS and then it should be possible to arm.