ArduPilot/ardupilot #17457: Build broken defining to use only one compass and/or GPS

giacomo892, 2021-05-16:
Generic plane omnibusf4pro build, defining 

```
define GPS_MAX_RECEIVERS 1
define GPS_MAX_INSTANCES 1
```

inside the hwdef.dat file and building:

```
lib/libArduPlane_libs.a(GCS_Common.cpp.4.o): In function `GCS_MAVLINK::try_send_message(ap_message)':
GCS_Common.cpp:(.text._ZN11GCS_MAVLINK16try_send_messageE10ap_message+0x3f0): undefined reference to `AP_GPS::send_mavlink_gps2_raw(mavlink_channel_t)'
collect2: error: ld returned 1 exit status

Build failed
```

Reducing compasses with: 

`define HAL_COMPASS_MAX_SENSORS 1`

Ends up with:

```
../../libraries/AP_Compass/AP_Compass_Calibration.cpp: In member function 'bool Compass::_start_calibration(uint8_t, bool, float)':
../../libraries/AP_Compass/AP_Compass_Calibration.cpp:56:9: error: '_priority_did_list' was not declared in this scope
     if (_priority_did_list[prio] != _priority_did_stored_list[prio]) {
         ^~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.

Build failed
```

giacomo892, 2021-05-16:
GPS bug addressed in #17459

giacomo892, 2021-05-18:
@bugobliterator Do you mind taking a look at the compass issue?

bugobliterator, 2021-05-18:
@giacomo892 why do you want to set this config `define HAL_COMPASS_MAX_SENSORS 1` ?

giacomo892, 2021-05-18:
Most of users just use one compass. Being able to customise this behaviour will allow to save some flash/ram and reduce the number of parameters to be downloaded each time

bugobliterator, 2021-05-18:
> Most of users just use one compass. Being able to customise this behaviour will allow to save some flash/ram and reduce the number of parameters to be downloaded each time 

Ok, I can create a PR to fix this issue, the main issue I see is that no board is going to have such config by default. Hence, the issue might get reintroduced in the future.

giacomo892, 2021-05-18:
> Ok, I can create a PR to fix this issue, the main issue I see is that no board is going to have such config by default. Hence, the issue might get reintroduced in the future.

Yeah I agree for the time being but in the future we will have a custom firmware builder so I guess it will come handy. Thanks

closed 2021-05-19 by tridge