ArduPilot/ardupilot #17475: AR_WPNav: Issue with WP_SPEED_MIN still exists

KimHyungSub, 2021-05-18:
## Bug report

Rover can go to reverse direction during AUTO flight mode when WP_SPEED_MIN is set.
#14840 reported this issue and #15951 fixed it. However, while I tested WP_SPEED_MIN parameter, I found that the rover did not progress towards a waypoint, but keep reversing at the WP_SPEED_MIN parameter value.

You can reproduce this issue as following steps:
1) param set WP_SPEED_MIN 2
2) Create a simple mission consisting of going to a waypoint and RTL
3) Keep repeating the mission

**Version**
master (0bb18a153c)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[V] Rover
[  ] Submarine

**Airframe type**
APMrover2

**Hardware type**
SITL

**Logs**
https://drive.google.com/file/d/1SRTpxmA8He_EO-hJqhGoaXzcfg3HVD6Q/view?usp=sharing

**Video**
https://drive.google.com/file/d/14NbUfYJ-VXKef57XYaoGUQWCKmJkaTwm/view?usp=sharing

rmackay9, 2021-05-18:
@KimHyungSub, thanks very much for this report.  Could you test using the Rover-4.1 beta and if possible report in the discussion here? https://discuss.ardupilot.org/c/ardurover/rover-41/158

The Copter-4.0.7 branch is only meant for Copter so it would not include fixes from master for Rover.

KimHyungSub, 2021-05-18:
@rmackay9 I tested WP_SPEED_MIN using Rover-4.1 beta and it seems that there is no issue. I would like to close this issue. Thanks!

closed 2021-05-18 by KimHyungSub

rmackay9, 2021-05-18:
@KimHyungSub, excellent, thanks very much for testing 4.1!