ArduPilot/ardupilot #17476: erratic SBUS Output / jitter reading

GlidLov, 2021-05-18:
## Bug report

**Issue details**

SBUS output is erratic, and somewhat PWM output also.
 I did some tests with a Teensy (arduino-like) programmed as SBUS decoder/encoder, connecting it to SBUS output of the Cube. I cannot read sbus stream correctly, it seems to read "glitches" and go 0 to all channels every few moments, very fast, like there's jitter into stream. If i connect to an X8R receiver or any other with sbus out, i can see a very stable reading of all channels without problems. Also i noticed that some accessories which uses PWM out, like landing gears, behave erratic when connected to Cube (they do randomic, fast open-close, like their PWM were jittery. No problems if i connect to a normal receiver. 
All 16 channels has been assigned, so there's no "0" outputs (All RCINs,  other than motors output etc). I tried to change SBUS stream refresh rates and i got a slightly better results at 100hz but no fix. Faster rates gives total unreadable stream.

**Version**

4.07 stable

**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
Cube (Black, Orange)

rmackay9, 2024-04-20:
I know this is quite an old report but we've made some changes on the SBUS output code in 4.5 and I wonder if you'd be willing to re-test with this newer version?

GlidLov, 2024-04-22:
i will as soon i have a quad ready