ArduPilot/ardupilot #17478: Camera settings causing Internal Error ISR Flood on designated feedback pin 0x2000000

stevenyoung17, 2021-05-18:
https://discuss.ardupilot.org/t/internal-error-0x2000000/70858/8
## Bug report

**Issue details**

Camera settings causing Internal Error ISR Flood on designated feedback pin 0x2000000
I have a Foxtech MAP-02 connected via AUX5 as relay trigger. I have CAM_FEEDBACK_PIN returning via hotshoe to AUX4. When camera is triggered via auto mission or controller. Internal error 0x2000000 is invoked and Quadplane goes into failsafe mode. 

**Version**
4.0.9 stable

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad Plane / Y3 design - Foxtech Nimbus
**Hardware type**
Cube Black
**Logs**
https://1drv.ms/u/s!Auv-5QlGCPQo-HpFw-RGA4SR_b29?e=KNtwjq

tridge, 2021-05-19:
this is definately a misconfig, with the same pin given multiple tasks. We need to detect that in pre-arm checks and warn user

IamPete1, 2021-08-16:
closing in favor of https://github.com/ArduPilot/ardupilot/issues/17465

closed 2021-08-16 by IamPete1