ArduPilot/ardupilot #17483: FS_OPTIONS: Continue if in Guided on RC - doesn't work

RM-FPV, 2021-05-18:
## Bug report

**Issue details**

I have FS_OPTIONS: Continue if in Guided on RC option set. If I select Guided mode and turn off the radio mode switches to RTL instead of  remaining in Guided.

**Version**
4.1.0-beta1

**Platform**
[+] Copter

**Airframe type**
X4

**Hardware type**
CC3D Revo

**Logs**
https://drive.google.com/file/d/1QPZzQlxzMsqB-_VcRl14LR54VbQSp2ai/view?usp=sharing

rmackay9, 2021-05-18:
Great, thanks for the report.  If possible going forward I'd prefer if the issues were first raises in the support forum here: https://discuss.ardupilot.org/c/arducopter/copter-41/159.  But thanks in any case and we can keep it open while I investigate.  Txs again.

rmackay9, 2021-05-18:
@RM-FPV,

The reason is that the GCS failsafe has been disabled.  The reasoning behind this design decision was that we didn't want vehicle to get stuck in Guided mode but with no connection to a GCS (or companion computer) to give instructions on where the vehicle should go.

So if you set FS_GCS_ENABLE = 1 then I think you'll find it works.

We could revisit this decision and/or add a pre-arm parameter check that if users have set FS_OPTION = 4 but not set FS_GCS_ENABLE = 1 we raise a pre-arm error.

RM-FPV, 2021-05-18:
Thanks! 
Ok, my real usage scenario is that I control the vehicle with the custom telemetry modem, and it's not connected to GS. It just sends mavlink commands. But I don't wont vehicle to Rtl with Rc FS. How can it be done without GCS?

rmackay9, 2021-05-18:
@RM-FPV,

If the custom telemetry modem sends HEARTEAT and/or RC_OVERRIDE messages then these will be interpreted as input from a GCS and will mean the GCS failsafe won't trigger.

The benefit of doing it this way is if that telem module fails the GCS failsafe can trigger and bring the vehicle home in RTL instead of the vehicle getting stuck.

rmackay9, 2021-05-19:
I'm going to close this issue but I'll be monitoring for replies in any case.

closed 2021-05-19 by rmackay9

RM-FPV, 2021-05-19:
Thank you very much for an explanation! I'll test it again and report.
Would be nice to add this info to the manual too.

Unfortunately for my particular flight routine this GS failsafe is not acceptable too. Is possible please to make it to work with battery FS instead of GS fs as an option?