ArduPilot/ardupilot #17497: UAVCAN ESCs: add support for reversing motor direction

rmackay9, 2021-05-20:
We should add support for reversing the direction of UAVCAN ESCs.  [See to-do in the code here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_UAVCAN/AP_UAVCAN.cpp#L478).

This was first raised as a request as part of the Rover-4.1.0 beta testing ([discussion](https://discuss.ardupilot.org/t/setting-up-skid-steer-with-uavcan-esc/71472))

jmachuca77, 2021-05-20:
I have this working already.... code needs cleanup though
https://github.com/ArduPilot/ardupilot/pull/15459

iq-fred, 2023-07-19:
This is a feature I am interested in adding. I added support for it on my own fork [over here](https://github.com/ArduPilot/ardupilot/compare/master...iq-fred:ardupilot:support-negative-dronecan-rawcommands) , and that seems to be working well enough during my testing with Rover, but I see now that there is an existing pull request as well.

I'm happy to work on pushing forward this feature, but I'm new to contributing to ArduPilot, so I had some questions:

- Is the preferred approach for me to make a new pull request from my fork pulling in my changes, or should I try to update the existing PR? It has been a couple of years since there was any movement on it at this point, so I'm not sure if its worth resurrecting that or just starting fresh.
- This question may be better off as its own issue, but I noticed on that previous PR and in the [DShot documentation](https://ardupilot.org/copter/docs/common-dshot-escs.html#reversible-dshot-escs) that Copter does not support reversible ESCs. I initially tested this in Rover which does support reversible ESCs, but I also tested in Copter out of curiosity and it did seem that it didn't really like negative throttles, the results were not as clean as on Rover. I would like to eventually be able to extend the use of reversible ESCs over DroneCAN to Copter. Can you give me any information on what would need to be done for Copter to support reverse thrust and reversible ESCs? Do you have any idea of what would happen if I tried to use a reversible ESC with DShot or DroneCAN in Copter's current state?

rmackay9, 2023-07-19:
Hi @iq-fred,

A new PR is probably best.

I think it would be best to first get the support for reversible DroneCAN motors in and the move onto a separate change to allow Copter to make use of them.  So the first stage would be driver changes and testing to ensure that Rover and Plane is improved and Copter (and perhaps other vehicles) are simply not affected.

Thanks!

iq-fred, 2023-07-20:
That sounds good, I'll focus on getting a pull request ready for changes to help out Rover and Plane first. Thanks!

iq-fred, 2023-07-24:
I've added a pull request to try and address this issue: #24386

closed 2024-09-11 by peterbarker