ArduPilot/ardupilot #17515: Copter: firmware version not displayed in startup banner

rmackay9, 2021-05-21:
While testing Copter-4.1.0 I've noticed that the firmware version no longer seems to be displayed in the ground station's messages window.  This makes it difficult to know which version is on the flight controller (you have to download a log to know).

[image]

This probably affects all vehicles but I've only confirmed it is an issue on Copter.

anbello, 2021-05-24:
I see the same problems on my Pixracer with Copter-4.1.0, if I change INS_FAST_SAMPLE from 1 to 0 I see firmware version on GCS again. If I swap again from 0 to 1 the message is no more displayed.

peterbarker, 2021-05-24:
On Sun, 23 May 2021, Andrea Belloni wrote:

> I see the same problems on my Pixracer with Copter-4.1.0, if I change INS_FAST_SAMPLE from 1 to 0 I see firmware version on GCS again. If I swap again from 0 to 1 the message is no more displayed.

Right.  We're overflowing the numbre of statustext messages that can be 
pending to be sent out (5).

We could increase the length of that buffer.

https://github.com/ardupilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS.h#L1038

(if I'm correct then you wouldn't see this on a H7-based board)

anbello, 2021-05-24:
> (if I'm correct then you wouldn't see this on a H7-based board)

correct

closed 2021-05-27 by peterbarker