ArduPilot/ardupilot #17552: Ground Effect Compensation broken in EKF3 (and probably EKF2)

rmackay9, 2021-05-25:
This PR https://github.com/ArduPilot/ardupilot/pull/15665 broke ground effect compensation by reducing the frequency with which the vehicle code calls EKF2 and EKF3's setTakeoffExpected() and setTouchdownExpected methods are called

Before the change the EKFs were always called, afterwards the early return means they are only called when the state is changed.  What was misunderstood however is that the EKFs have a one second timeout meaning that if they are not called continuously with "true" they revert to "false".

peterbarker, 2021-05-26:
Looks like this means we utterly broke ground-effect-compensation for quadplane, too, as it only ever sets this flag to true, meaning it will time out once within EKF2 and EKF3 and then never be set down into there again.

priseborough, 2021-05-26:
We could change the EKF to use the last set status and not apply the timeout.

rmackay9, 2021-05-26:
@priseborough

this crossed my mind as well.. how dangerous do you think it is if the flags were accidentally left on too long?  Do you think it could lead to a fly-away or climb-away?  The concernt that we could introduce a bug in the future that leads to the flag not being reset back to false (ever) is  the only thing stopping me from recommending we remove the EKF's timeout.

Removing the timeout would make the code easier to understand.

priseborough, 2021-05-27:
@rmackay9 It wouldn't lead to a flyaway if left on too long, but there could be up to 5m of height error develop as time progresses with the rate of error growth depending on the accuracy of the GPS vertical velocity. We can modify the copter code to ensure that it has a timeout which should be easy to do given the logic is contained in a single function.

PR #17575 (2021-06-02): Fix ground effect compensation
Closes #17552 

Apart from the autotest I've added here, I've manually reviewed the logs produced by the autotest to ensure the flag continues to be set as appropriate.

I've also tested quadplane before/after this patch.  Before the patch the ground effect compensation was only turned on for a second (the predicted broken behaviour), after the fix it remained on for ~20 seconds while the vehicle was `QLAND`ing.

The solution here effectively decimates the messages we pass down to the EKFs.  I would have actually preferred to eliminate the timeout of the state do [... cut]
closed 2021-06-02 by rmackay9