ArduPilot/ardupilot #17559: Impossible to use BLHeli bidir soft mode with master

shellixyz, 2021-05-25:
## Bug report

**Issue details**

It is impossible to use BLHeli's bidir soft mode since `RCOutput::set_reversible_mask()` switches the ESC to 3D mode overiding the mode set in BLHeli suite if the output is marked as reversible

**Version**

current master branch

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

shellixyz, 2021-05-26:
I guess what could be done is add a param to chose whether AP should force 3D mode by itself or leave the current ESC setting

andyp1per, 2021-05-26:
Ok so I guess currently you can leave SERVO_DSHOT_ESC = 0 and it will work but then you lose the other dshot command features.
I can ask Steffen about adding soft bi-dir. You are right that adding a parameter probably makes sense, or maybe an option bit to 3DMASK.

shellixyz, 2021-05-26:
> Ok so I guess currently you can leave SERVO_DSHOT_ESC = 0 and it will work but then you lose the other dshot command features.

Yeah, no DSHOT commands means no bidir with DSHOT at all and also less importantly for me at least no audio notifications and no LED control

andyp1per, 2021-05-26:
> No DSHOT commands means no bidir with DSHOT at all

I don't understand what you mean here? dshot commands are not required for bi-directional dshot and soft 3D can be set on the ESC?

shellixyz, 2021-05-26:
> > No DSHOT commands means no bidir with DSHOT at all
> 
> I don't understand what you mean here? dshot commands are not required for bi-directional dshot and soft 3D can be set on the ESC?

My bad, you are right, DSHOT commands are not needed for bidir/revthrust