ArduPilot/ardupilot #17572: SITL/Linux build does not support MAV_CMD_PREFLIGHT_REBOOT_SHUTDOWN

patrickelectric, 2021-05-26:
## Bug report
SITL and Linux builds does not reboot/restart when receiving `MAV_CMD_PREFLIGHT_REBOOT_SHUTDOWN`

**Version**
master

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All

**Hardware type**
SITL/Linux

khancyr, 2021-05-26:
Hum how do you use the command. It definitely works on rover and copter with mavproxy.
That is using pymavlink and the same message : https://github.com/ArduPilot/pymavlink/blob/118bcc4bfc2998c3ae825c6654f039cdf644db2d/mavutil.py#L765

patrickelectric, 2021-05-26:
Hi @khancyr, sorry for that, I was only able to replicate the issue for a Linux target and not SITL.

rmackay9, 2021-05-26:
@patrickelectric,

The vehicle was disarmed of course?  I think it rejects reboot commands if the vehicle is armed.

khancyr, 2021-05-27:
I don't have Linux board, but it shouldn't be hard to implement the reboot like in SITL. The only issue may be with the RT stuff as we will to ensure that we close everything correctly before exiting

patrickelectric, 2021-05-27:
It just ends the program:
[image]

Thanks @khancyr, I'll investigate it

patrickelectric, 2021-05-27:
The comment is a bit weird, but it should work, any idea if this is the correct behaviour ?
https://github.com/ardupilot/ardupilot/commit/ce4c4c43dcf642c96f08eabd84ccc14594909723

juvinski, 2021-05-27:
Hi,

to use the reboot, I have a service on the systemd, once the process die - and this is the behavior of the reboot command, the systemd start the ardupilot program again.

khancyr, 2021-05-27:
@patrickelectric if that is exit then reboot isn't implemented. Copying what is in SITL should be the first step (the hal and scheduler reboot logic) as SITL is Linux based. Then we should check that we don't lock thing on RT or other side things.

patrickelectric, 2021-05-27:
That's my plan @khancyr, I'm right now traveling, will do that when possible if no one does that first.

rafaellehmkuhl, 2021-08-23:
I can confirm the following behavior:

- Both HAL_SITL and HAL_LINUX do not support shutdown command (param1=2);
- HAL_SITL correctly reboots on reboot command (param1=1);
- HAL_LINUX performs a shutdown (process dies) on reboot command (param1=1);