ArduPilot/ardupilot #17579: Move test_onboard_logging_generation outside autotest.py

khancyr, 2021-05-27:
## Feature request

**Is your feature request related to a problem? Please describe.**
Move test_onboard_logging_generation outside autotest.py to all to validate logger documentation before running the full autotest suite. Ideally, on CI it should run on build_ci.sh with param_parse.py to allow to validate soon that the documentation is correct and not wait that all test pass to trigger a documentation failure...

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

shiv-tyagi, 2021-09-20:
Hey! I want to work on this issue. I am new to autopilot. Can anyone elaborate a bit about what this issue is talking about and how can I approach to solve it.

tzarjakob, 2023-03-07:
Is this issue still open?