ArduPilot/ardupilot #17583: Copter: implement "soft landing" in Loiter (and other similar modes)

rmackay9, 2021-05-27:
This enhancement request is implement a "soft landing" by limiting the pilot controlled yaw, horizontal and vertical velocity just before landing based on the distance returned by a downward facing lidar.

In modes where the pilot controls the climb rate (Loiter, PosHold, AltHold, etc) the descent rate could be limited so that the vehicle does not hit the ground at more than 0.5m/s.

In modes that allow the pilot to control the horizontal velocity (Loiter, Land, final stages of RTL, SmartRTL, Auto-Land), these velocities could also be limited so that they become zero just before touch down.  They could also be limited at higher altitudes to ensure that the velocity can be reduced to zero before touchdown based on the maximum descent rate and maximum horizontal acceleration.  Note that this might actually involve limiting the horizontal velocities even when the vehicle is hovering or climbing (at low altitude) because the pilot could pull the stick down at any moment.  Alternatively the descent rate could be limited until the horizontal velocity has been reduced (i.e. the pilot pulls down and this triggers limiting of the horizontal velocity but the desired descent rate remains at zero until the horizontal velocity has actually been reduced).

The desired yaw rate could similarly be limited to ensure the vehicle is not rotating just before touchdown (or even soon after touchdown).

This request stems from [Biswajeet_prusty's discussion in the Copter-4.1 category](https://discuss.ardupilot.org/t/arducopter-feature-request-land-asap-when-throttle-stick-is-zero/71703).

davidbitton, 2021-05-30:
@rmackay9 if FS_THR_VALUE is set to just above 1000, and FS_THR_ENABLE is set to "Enabled always Land", then wouldn't pulling the throttle to zero on the TX force Land mode?

rmackay9, 2021-05-30:
@davidbitton,

Yes, that would work but I'm not sure that's a great overall solution thought to this enhancement request because it would mean the only way the user could recover altitude control would be to switch the vehicle back to Loiter (or whichever mode they were using).  Also this would mean the throttle failsafe couldn't be set to RTL which is probably the best setting for many users.

davidbitton, 2021-05-31:
@rmackay9,

If the throttle was moved above the failsafe point, won't Arducopter revert to the prior flight mode?

rmackay9, 2021-05-31:
@davidbitton, No I don't think so.  Once the failsafe causes the vehicle to switch modes it won't go back on its own.  The only way it might switch back to the pre-failsafe flight mode is if the receiver somehow causes the flight mode switch to be moved and that recovery happens *after* the throttle has recovered.

shiv-tyagi, 2022-01-26:
@rmackay9 I am very much interested on working on this issue. I will try doing some work on this one.