ArduPilot/ardupilot #17593: CRSFv3 Support

Johnex, 2021-05-28:
Hi!

Are there any plans to upgrade to the newly released CRSFv3 protocol?

New features:

Support 24 channels (increased from 16)
CRC check improvements
Support sending subset of maximum number of channels
Support configuration of resolution of the RC channel data
Baudrate negotiation
New link statistic frames

For reference here is the TBS commit into Betaflight:

https://github.com/betaflight/betaflight/pull/10675/files

andyp1per, 2021-05-28:
Sure if @tbs-fpv give us some support - my contact at TBS - Remo - has left and the other contact I have is not responding

Johnex, 2021-05-28:
From what I understand they are upgrading the entire protocol from OpenTX all the way to the receiver/fc, so it might be required in the future to support their products, so I hope they do offer some support for you guys 👍

amilcarlucas, 2021-05-28:
transparent Mavlink 2.0 support? Now that would be a nice feature!

tbs-trappy, 2021-06-11:
> transparent Mavlink 2.0 support? Now that would be a nice feature!
this is in the works, independently from the CRSF V3 implementation. email me (trappy at gmail) if you'd like to help with the testing efforts.

tbs-trappy, 2021-06-11:
> From what I understand they are upgrading the entire protocol from OpenTX all the way to the receiver/fc, so it might be required in the future to support their products, so I hope they do offer some support for you guys 👍

the protocol is backwards-compatible and we don't plan on dropping legacy support anytime soon, so there is no immediate need to upgrade. obviously, new is (almost) always better :)

tbs-trappy, 2021-06-11:
> Sure if @tbs-fpv give us some support - my contact at TBS - Remo - has left and the other contact I have is not responding
who is your other contact at TBS?

Johnex, 2021-07-08:
@tbs-trappy Where you guys and the Ardupilot guys able to make a plan for adding CRSFv3 to Ardupilot?

andyp1per, 2021-07-08:
Yeah I have kit and firmware - now just need to find the time to do it!

Johnex, 2021-07-09:
@andyp1per Let me know if you need any help testing 👍

andyp1per, 2021-07-10:
First cut, completely untested: https://github.com/ArduPilot/ardupilot/pull/17990

Johnex, 2021-07-10:
@andyp1per Code looks really nice, its a great start. Will you implement the remaining things like the command frames for baudrate negotiation? I also didn't see if there was a limit currently for the number of channels, or is that handled in the packed frames?

andyp1per, 2021-07-10:
I probably won't do the baud rate negotiation - this presumably is only useful on chips using UARTS without DMA. We have recently discovered that on H7, FIFO gives us sufficient speed, so the complexity is probably not worth it.

I don't think my implementation is correct because the number of channels is totally dependent on the frame size, so I think I will probably lift the beta flight code and use that.

Johnex, 2021-07-10:
Yeah that was why I asked, I saw you were using the max frame size, but more channels could be pending in the next frame so you need to check the actual payload total size and get all the channels based on that and upcoming frames. If you lift the betaflight code check the later commits after the main one, they fixed some issues.

andyp1per, 2021-07-14:
@Johnex this appears to be working now for me, feel free to give it a try - I had to implement baud rate negotiation as that is how the RX tells it is using v3

Johnex, 2021-07-14:
@andyp1per I saw your changes, looks great 👍 I am currently waiting for the firmware after Crossfire firmware 6.03 (as there were some issues and TBS will drop a new fix version), should be a few days hopefully the end of the week when that drops, then i can flash all my devices for v3.

andyp1per, 2021-07-14:
Ok, I have been testing with 6.01 and have to say its a bit flaky. I couldn't get later versions to bind.

Johnex, 2021-07-15:
In TBS Agent X, the changelog for 6.06 says this:
"This release is considered early beta and should only be installed if you have a TBS Crossfire or TBS Tracer Sixty9 and want to use it. There will be a following release replacing this one in the coming week."

I have never seen such a message before, so TBS probably found some serious issues, worth waiting for the new fix. The firmware was a bit rushed for their Sixty9 release, so it's understandable some issues slipped the cracks as well.

tbs-trappy, 2021-07-15:
Yes, there was a known bug affecting binding in Crossfire which we don't recommend "installing" on your units. However, we are currently alpha-testing 6.07 and it should replace 6.06 tomorrow. That one is considered stable.

Johnex, 2021-07-15:
@tbs-trappy 👍 have a wonderful day, see you in the livestream tomorrow? 😛

tbs-trappy, 2021-07-15:
ha, let's see what time allows :)

andyp1per, 2021-07-20:
@Johnex any joy testing?

Johnex, 2021-07-21:
@andyp1per The new crossfire firmware hasn't dropped yet, still waiting for it 😭

Johnex, 2021-07-29:
@andyp1per I tested your new code on a Holybro Durandal, works fine as far as I can tell. I don't have an oscilloscope or advanced tools to check any performance apart from just the basic Mission Planner outputs. Before merging someone will probably want to test the changes if they have better tools for it. Maybe  @tbs-trappy can see if there are any issues?

andyp1per, 2021-08-04:
Fixed by https://github.com/ArduPilot/ardupilot/pull/17990

closed 2021-08-04 by andyp1per