ArduPilot/ardupilot #17617: Quadplane: Battery Failsafe in Q modes causes transition to fixed-wing RTL.

cheintz, 2021-06-01:
## Bug report

When a quadplane encounters a battery failsafe while in a Q mode (QLoiter, etc) and very near the home location with, it transitions to RTL mode. This is not desired and can be quite dangerous if it happens near the ground and is not caught in time.

This occurred with the following relevant parameters:
BATT_FS_CRT_ACT = 4 
BATT_FS_LOW_ACT = 1 
Q_OPTIONS = 9 ( Level Transition and Respect Takeoff frame types)
Q_RTL_MODE  = 0

I think it would make sense for the actions to depend on at least the current mode (Fixed-wing vs. Quad).
A useful minimum implementation would be to have different BATT_FS_LOW and CRT actions for Fixed-Wing and Quad flight modes.  I might set my 4 ACT parameters like this:
BATT_FS_Q_LOW_ACT: Q_RTL (hopefully a smart RTL) and Q_LAND at home 
BATT_FS_Q_CRT_ACT: Q_LAND immediately.
BATT_FS_FW_LOW_ACT: Plane RTL
BATT_FS_FW_CRT_ACT: Plane RTL

It's possible the functionality in 4.1 and beyond could make this more advanced, such as having numerous landing sites so that the plane could fly as fixed-wing and perform something fairly close to a conventional approach, transitioning only at the last second to avoid pulling too much current/charge from the battery. 

**Version**
4.0.7

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
4+1 Quadplane (Foxtech Loong)

**Hardware type**
CUAV V5+

**Logs**
(Yes, I Force-armed through the BATT_ARM_VOLT pre-arm check, but only wanted to do a little hover).

[2021-05-31 15-03-08.zip](https://github.com/ArduPilot/ardupilot/files/6573092/2021-05-31.15-03-08.zip)

pompecukor, 2021-06-04:
This was expected behavior. You should have set Q_RTL_MODE = 1
That is the main point of that parameter. To control what happens when RTL is inside home radius.

cheintz, 2021-06-05:
I find it useful to have Plane RTL available if already in plane mode. For example: To upload a new mission, to address GCS/RC connection issues,