ArduPilot/ardupilot #17620: GPS-for-Yaw improvements

rmackay9, 2021-06-01:
This is a consolidated list of enhancements that would make GPS-for-yaw work better and/or ease the setup for users:

- [ ] Implement GPS-for-yaw via UAVCAN.  This would allow connecting two CAN GPSs to the autopilot without adding the extra CPU and I/O overhead which comes from using Serial ports because the CAN messages would flow from the Rover to the Base (through the autopilot) without the autopilot having to parse them.
- [ ] Enhance EKF3 to consume yaw separately from position and velocity (currently they must be consumed from the same GPS at the same time).  This would both improve accuracy (position and velocity are most accurate from the "base" while yaw must come from the "rover") and avoid the GPS switching we see when we lose the yaw estimate from the GPS (when this happens we switch to the "base" because it gives a better position and velocity estimate.
- [ ] Add a GPS_OPTIONS flag to for the GPS driver to use the baud rate specified by the SERIALx_BAUD parameter instead of using the GPS driver's autobauding
- [ ] Enhance the GPS-for-yaw wiki page to explain how to connect the two GPSs together directly

MallikarjunSE, 2021-06-01:
I can test GPS for yaw on 2×AP_periph F9P devices. If it helps!

IamPete1, 2022-09-10:
@rmackay9 I think at least a few of these have been done?

rmackay9, 2022-09-11:
@IamPete1,

Yes, I think these have largely been done.  For the 2nd last one we only have a very specific "Use 115200" option but that's the one that was really required.  The very last one is really a wiki thing so I think I'll close this now.  txs

closed 2022-09-11 by rmackay9