ArduPilot/ardupilot #17623: Arduplane latest 4.1 BLHTEMP does not work in DJI MSP OSD

DarkSoulRL, 2021-06-01:
#### Issue details
 BLHTEMP does not work in DJI MSP OSD

#### Version
Arduplane 4.1 
From 2021-05-12 it is not displayed on OSD.
Before this date, the temperature was displayed, but it did not change.

#### Platform
[ X] Plane
#### Airframe type
Flying wing

#### Hardware type
Matek H743 slim

ESC telemetry works, temperature, voltage, current and RPM are recorded in log normally.

[image]

shellixyz, 2021-06-02:
#17569 fixes that

DarkSoulRL, 2021-06-02:
@shellixyz 
I tried compiling and installing your fork https://github.com/shellixyz/ardupilot/tree/fix/dji_fpv_osd_temperature_and_rpm.
BLHTEMP is displayed, but always shows 1 celsius degree.

shellixyz, 2021-06-02:
Make sure you have MSP_OPTIONS set to 2.

DarkSoulRL, 2021-06-02:
The same doesn't work.

DarkSoulRL, 2021-06-02:
When turned on, the correct temperature flashes, then again 1 degree Celsius.

shellixyz, 2021-06-02:
That's quite strange. I tested it and it is working for me. Usually when it displays 1 or 2 it is because it is using the old code (MSP_OPTIONS bit 1 at 0) so it is displaying the number of motors

rmackay9, 2021-06-02:
FYI @andyp1per, @tridge 

As a side note this really is a support issue so ideally it should be in discord but we don't have a Plane-4.1 discussion category yet so I'll leave this open for a couple more days 'cuz we also don't want to shut down this productive conversation.

DarkSoulRL, 2021-06-03:
Hello. @shellixyz , Please compile the target MATEKH743 for me. Maybe I'm doing something wrong.

shellixyz, 2021-06-03:
> Hello. @shellixyz , Please compile the target MATEKH743 for me. Maybe I'm doing something wrong.

Here you go [MatekH743-bin-plane-346b041022.zip](https://github.com/ArduPilot/ardupilot/files/6591630/MatekH743-bin-plane-346b041022.zip)

DarkSoulRL, 2021-06-03:
BLHTEMP parameter is not displayed on OSD :(
Here's my config:
[2021-06-03-master-fix.zip](https://github.com/ArduPilot/ardupilot/files/6591798/2021-06-03-master-fix.zip)

shellixyz, 2021-06-03:
@DarkSoulRL Is that working for you ?

[MatekH743-plane-bin.zip](https://github.com/ArduPilot/ardupilot/files/6592056/MatekH743-plane-bin.zip)

DarkSoulRL, 2021-06-03:
Yes, It works!

DarkSoulRL, 2021-06-04:
What have you fixed? I don't see any changes in the code.

shellixyz, 2021-06-04:
`#ifdef HAL_WITH_ESC_TELEM` to `#if HAL_WITH_ESC_TELEM`

DarkSoulRL, 2021-06-05:
There is a problem. In a real flight, about a minute after the arm, the temperature sensor image update stops :( I made 4 flights today - this happens every time.
Also, the rate of update of the mAH consumption has become very rare.

DarkSoulRL, 2021-06-05:
I checked the logs ... The temperature did not change in them either. Strange, this has not happened before.

DarkSoulRL, 2021-06-05:
I don’t know why, but often when the power is turned on, the regulator does not turn on as DSHOT
05.06.2021 17:12:28 : RCOut: PWM:1-13

SERVO_BLH_MASK ,1
SERVO_BLH_OTYPE ,5

shellixyz, 2021-06-05:
I haven't encountered these issues. They are unrelated to this report's topic please discuss them in Discord or the forum.

DarkSoulRL, 2021-06-05:
Is it a good idea to enable "Auto Telemerty = ON" in BLHeli configurator?
In this case, telemetry is available even if ESC is in PWM mode.

DarkSoulRL, 2021-06-13:
I flew on your firmware several times. The problem is that my DSHOT is unstable, and telemetry accordingly. In order for everything to work as it should, you need some kind of dances with a tambourine. Turn on the power supply of the flight controller via USB, then connect the battery. Restart the flight controller by command with MP, then without disconnecting the USB, reconnect the battery.
I found out that if you disable the CAN bus, then everything works stably. But I have a GPS Matek M8Q CAN connected to it. Here I described https://github.com/ArduPilot/ardupilot/issues/17688

(sorry - google translate)

closed 2021-07-06 by tridge

DarkSoulRL, 2021-07-06:
I compiled and installed today's master. And now ESCTEMP always shows 0.
MSP_OPTIONS = 2.

shellixyz, 2021-07-10:
> I compiled and installed today's master. And now ESCTEMP always shows 0.
> MSP_OPTIONS = 2.

Small mistake, it displays motor temperature instead of ESC temperature. Fixed by #17993

PR #17569 (2021-07-06): AP_MSP: fix DJI FPV temperature and RPM
Version 1.00.06 (latest at the time this PR is submitted) of DJI firmware is not using the standard MSP_ESC_SENSOR_DATA. In fact no version is AFAIK.

With MSP_OPTIONS = 2 (DJI Workarounds) the temperature and RPM are displayed correctly. Since only the one temperature and one RPM can be displayed what is sent to the Air Unit/OSD is the maximum temperature and average RPM across all ESCs.

Closes #17623