ArduPilot/ardupilot #17640: Square root controller

hmedehuss, 2021-06-02:
Hello guys, i beleive there is an error when calculating the linear_dist in the square root controller which could lead to discontinuities when calculating the desired target.
The linear distance in case both the P and second order limit have been defined is calculated by:

```javascript
  const float linear_dist = second_ord_lim / sq(p);            (*)
```
However i think it should be 

```javascript
const float linear_dist = second_ord_lim * sq(p);              (**)
```

However, when the condition -linear_dist<error<linear_dist is achieved , the equation (*) is not consistent with the correction rates calculated outside this condition.

rmackay9, 2021-06-03:
@hmedehuss,

This is probably a question that @lthall has to answer.  Maybe try changing it and see how it performs?  I'd be a little surprised if we made such a fundamental mistake but Leonard will know.

peterbarker, 2021-06-03:
@hmedehuss can you suggest a test we could add to our unit tests to show this problem?

https://github.com/ArduPilot/ardupilot/pull/17648/commits

... realised I hadn't actually PR'd that old commit 'til just now.

lthall, 2021-06-03:
I am sorry but it isn't clear to me what you are saying is not right. Can you go into a little more detail to highlight the issue?

lthall, 2021-07-21:
No reply and no sign of a problem.