ArduPilot/ardupilot #17641: AP_OSD: BLH RPM/TEMP/AMPS: nothing is displayed if there is no ESC connected on the first output

shellixyz, 2021-06-02:
## Bug report

**Issue details**

AP::esc_telem().get_{temperature,rpm,current}(0) returns false if there is no ESC connected on the first output so the OSD doesn't display anything in this case. Seems like the index is taken as output index instead of ESC index. So if the motor is wired on output 3 the telem.get_xxx(0) functions return false.

**Version**

Current master

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

andyp1per, 2021-06-03:
Yes I was thinking about this the other day, we need something like get_first_active_esc()

shellixyz, 2021-06-03:
I'm assuming then it is the intended behaviour (index argument as output index). I wasn't sure about that but now that it is clear I think the best fix is to use the new ESC telem methods I added in #17569 to display the highest temperature and RPM average. I will wait for it to be merged to PR the change.