ArduPilot/ardupilot #17642: AP_Scheduler: tasks scheduled prematurely every ~163 seconds with a 400 Hz main loop

ntamas, 2021-06-02:
## Bug report

**Issue details**

The scheduler class executes all tasks before their time is due every ~163 seconds on ArduCopter 4.x. This can easily be tested in the SITL by setting `SCHED_DEBUG` to 3 and then watching for `Scheduler slip task` messages on the (simulated) UART; every ~163 seconds there is a stream of messages like this:

```
Scheduler slip task[0-rc_loop] (4294901764/4/130)
Scheduler slip task[1-throttle_loop] (4294901768/8/75)
Scheduler slip task[2-update_GPS] (4294901768/8/200)
Scheduler slip task[3-update] (4294901762/2/160)
...
Scheduler slip task[46-terrain_update] (4294901776/40/100)
Scheduler slip task[47-update] (4294901776/40/75)
Scheduler slip task[48-update] (4294901776/80/100)
...
```

Note the incorrect `dt` value, which is very close to 2^32 in all cases. This is actually the root cause of the issue; `dt` was changed from an `uint16_t` to an `uint32_t` between ArduCopter 3.x and 4.x, so it no longer handles the overflow of the 16-bit tick counter correctly. (At least not with my C compiler).

You may also notice the bug from the `PERF` warning messages in the MAVLink stream if you don't have access to the debug console; every 163 seconds there will be a `PERF` message where the number of loops executed since the last message will not be around 4000 (as expected for a 400 Hz loop every 10 seconds) but 1536; note that 65536 - 16*4000 = 1536.

**Version**

ArduCopter 4.0.7, but it affects all versions from 4.0.0. Versions in the 3.x series are not affected.

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

Not applicable.

**Hardware type**

Tested on a Pixhawk1 and on a CubeBlack with stock firmware; same issue, but the reproduction is easier in the SITL.

**Logs**

Not applicable.

ntamas, 2021-06-02:
I think [this](https://github.com/ArduPilot/ardupilot/commit/79acaeaef1adca79fcac5aeb317f0a4aa70c97f8) is the commit that introduced the problem.

rmackay9, 2021-06-02:
This looks like a good find, thanks!

peterbarker, 2021-06-03:
On Wed, 2 Jun 2021, Tamás Nepusz wrote:

> The scheduler class executes all tasks before their time is due every ~163 seconds on ArduCopter 4.x. This can easily be tested in the SITL by setting SCHED_DEBUG to 3 and then watching for [...]

Great find!

Since you've studied the problem a bit - do you have a suggested solution?

ntamas, 2021-06-03:
The easiest is to switch the type of `dt` and `interval_ticks` in `AP_Scheduler::run()` (see the commit I referenced above for context) back to `uint16_t` from `uint32_t`; however, there must have been a reason why they were switched from `uint16_t` to `uint32_t` in the first place and I'm afraid I'm not seeing the whole picture here. I am happy to submit a PR that changes the type; I tested it on my CubeBlack (sitting on my desk) and it seems to have resolved the problem, but maybe it would be better to see what @tridge thinks as he made the original commit that changed the type to `uint32_t`.

tridge, 2021-06-08:
PR would be welcome, thanks!

ntamas, 2021-06-08:
@tridge PR sent, thanks!

commit c544f3dd (2021-06-10): AP_Scheduler: corrected tick counter overflow handling, fixes #17642
closed 2021-06-10 by peterbarker