ArduPilot/ardupilot #17645: SITL: tri model doesn't work

rmackay9, 2021-06-03:
At some point the multicopter Tri frame type in SITL stopped working.  It may be just tuning but in any case, we really should get this working again.

autotest has a full list of vehicle frames that don't fly in SITL here: https://github.com/ardupilot/ardupilot/blob/master/Tools/autotest/arducopter.py#L6562

Note that this does not mean that the real vehicles don't fly.  It most likely just means the SITL parameter files are incomplete for these frames.

tridge, 2021-06-08:
fix here: https://github.com/ArduPilot/ardupilot/pull/17695

hendjoshsr71, 2021-06-18:
Closing. Fix merged in #17695

rmackay9, 2021-06-28:
This issue may not be fixed at least according to this discussion: https://discuss.ardupilot.org/t/fly-away-in-land-mode/73001

rmackay9, 2021-06-28:
I've investigated a bit and it is not the same issue as before so closing.  I'll create a new issue.

closed 2021-06-28 by rmackay9