ArduPilot/ardupilot #17656: BLHeli Telemetry and Batt issues

Alexis-Stu, 2021-06-04:
**Issue details**
Battery (BLHELI_Telemetry sum) reports the correct voltage but when amps are drawn, the percentage begins to fall too fast given the parameters and amps drawn.

ESC telemetry values appear to be recorded in one message in the log unlike the stable firmware that records a separate instance for each ESC. This may be causing a current draw issue that is reporting far too many mAh consumed compared to the amps drawn.

**Version**
4.1.0 Latest

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Twin engine electric

**Hardware type**
Pixracer Pro, Holybro BLHeli_32 ESC's.

**Logs**
https://1drv.ms/u/s!AunBK4rE1ZDChI8fAfDa9Yui6xwh9w?e=UKh1IS

andyp1per, 2021-06-04:
@Alexis-Stu your log show multiple telemetry values:

[image]

can you explain a bit more what you think the problem is? Note that the consumption figure comes straight from BLHeli - we do not do any interpretation on it

Naterater, 2021-06-04:
Look at the BAT.CurrTot message. It is not the sum of the ESC.Ctot messages. In just a few minutes at 0 throttle, it shows 3500mAh consumed. I also see DroneePlotter doesn't handle the instances very well. 

[image]

andyp1per, 2021-06-04:
Right, but this comes direct from BLHeli so I would say is a BLHeli issue

Naterater, 2021-06-04:
No. BLHeli is telling us that a total of 10mAh have been consumed. The BAT.CurrTot tells us that 3500mAh have been consumed.

This works just fine in 4.0.9.

andyp1per, 2021-06-04:
Oh sorry, now I see what you are saying. How do you have the battery monitor configured? Is it set to ESC?

Naterater, 2021-06-04:
Yes. BATT_MONITOR is set to option 9

Naterater, 2021-06-04:
Possible related issue I'm seeing; on some boots, the BATT doesn't even recognize the BLHeli voltage readings. It stays at 0 even though ESC 5 and 6 are showing voltage through Mission Planner.

andyp1per, 2021-06-04:
Ok I see what is going on here: this should fix https://github.com/ArduPilot/ardupilot/pull/17657

rmackay9, 2021-06-04:
I've added this to the 4.1 issues list .. looks like we have a fix in the works so hopefully we can include this in beta4

andyp1per, 2021-06-09:
Hi @Naterater or @Alexis-Stu - are you able to try https://github.com/ArduPilot/ardupilot/pull/17657 ? I do not have ESCs that have current so cannot. I can do a build for you if you need.

In particular can you check that battery reset works correctly:

[image]

Naterater, 2021-06-09:
Please send a mRo Pixracer Pro build and we'll get it tested today.

andyp1per, 2021-06-09:
@Naterater:
[arducopter.zip](https://github.com/ArduPilot/ardupilot/files/6625277/arducopter.zip)

Naterater, 2021-06-09:
@andyp1per, Sorry, plane please

andyp1per, 2021-06-09:
Done

andyp1per, 2021-06-15:
@Naterater any joy?

Naterater, 2021-06-15:
@andyp1per, yes, that appears to have solved the issue.

andyp1per, 2021-06-15:
@Naterater super! And battery reset works as expected?

Naterater, 2021-06-15:
As far as I could tell, yes. We did reset it twice and it performed nominally.

andyp1per, 2021-06-15:
Thanks @Naterater

closed 2021-06-20 by andyp1per