ArduPilot/ardupilot #17688: Arduplane 4.1 DSHOT is not always enabled if CAN bus is used.

DarkSoulRL, 2021-06-07:
**Issue details**
When power is applied or rebooted, DSHOT does not always turn on if CAN bus is used.

**Version**
Arduplane 4.1 latest

**Platform**
[X ] Plane

**Airframe type**
Flying wing

**Hardware type**
Matek H743 slim, also tested on fresh install Pixhawk 2.4.8

**Logs**

**How to repeat bug**
1. Install latest firmware and reset to defaults.
2. Enable DSHOT: 
    SERVO_BLH_OTYPE, 5
    SERVO_BLH_MASK, 1
3. check messages on boot:
07.06.2021 15:22:30 : IMU0: fast sampling enabled 8.0kHz/2.0kHz
07.06.2021 15:22:30 : **RCOut: DS300:1-2 PWM:3-13**
07.06.2021 15:22:30 : MatekH743 0029002B 33395105 33383930
07.06.2021 15:22:30 : ChibiOS: 08877972
07.06.2021 15:22:30 : ArduPlane V4.1.0dev (346b0410)
07.06.2021 15:22:28 : Airspeed 1 calibrated
4. reboot/power on/power off 5-7 times,  and make sure always **RCOut: DS300: 1-2 PWM: 3-13**
5. enable and configure the CAN bus according to the instructions from the Matek website for the M8Q CAN
CAN_D1_PROTOCOL -> 1
CAN_P1_DRIVER -> 1
GPS_TYPE -> 9
6. check messages on boot:
07.06.2021 15:42:02 : Airspeed 1 calibrated
07.06.2021 15:42:01 : EKF3 IMU1 buffs IMU=17 OBS=7 OF=16 EN:16 dt=0.0200
07.06.2021 15:42:01 : EKF3 IMU0 buffs IMU=17 OBS=7 OF=16 EN:16 dt=0.0200
07.06.2021 15:42:01 : Flight mode = 1
07.06.2021 15:42:01 : Failsafe. Short event on: type=1/reason=3
07.06.2021 15:42:01 : Throttle failsafe on
07.06.2021 15:42:00 : GPS 1: specified as UAVCAN1-112
07.06.2021 15:42:00 : AHRS: DCM active
07.06.2021 15:42:00 : ArduPilot Ready
07.06.2021 15:42:00 : Airspeed 1 calibration started
07.06.2021 15:42:00 : Barometer 2 calibration complete
07.06.2021 15:42:00 : Barometer 1 calibration complete
07.06.2021 15:41:59 : Calibrating barometer
07.06.2021 15:41:59 : IMU0: fast sampling enabled 8.0kHz/2.0kHz
07.06.2021 15:41:59 : **RCOut: PWM:1-13**
07.06.2021 15:41:59 : MatekH743 0029002B 33395105 33383930
07.06.2021 15:41:59 : ChibiOS: 08877972
07.06.2021 15:41:59 : ArduPlane V4.1.0dev (346b0410)
7. reboot/power on/power off 5-7 times,  and see that DSHOT is almost always unused.

DarkSoulRL, 2021-06-07:
If DSHOT does not turn on, ESC telemetry does not work, regulator temperature is not displayed, RPM filter does not work either.

magicrub, 2021-06-07:
Please confirm that you're aware that both CAN_D1_PROTOCOL and CAN_P1_DRIVER require reboots to take effect.

DarkSoulRL, 2021-06-08:
Yes of course. I reconnect the power to connect the M8Q CAN GPS after setting the parameters for the CAN bus.

DarkSoulRL, 2021-06-08:
Before that, I had the ArduPlane V4.1.0dev build (444db8f6), and here DSHOT and CAN bus worked fine.

sh4jonny, 2021-06-30:
I confirmed this issue, but i even didn't have a CAN. trying on matek f405-wing and 4.1.0 b2 with crossfire configured

IamPete1, 2021-08-08:
@andyp1per Maybe some dma contention issue?

andyp1per, 2021-08-09:
@IamPete1 thanks for the reminder - I should try this out

DarkSoulRL, 2021-08-09:
I am currently using arduplane beta4 with wiks from here https://github.com/ArduPilot/ardupilot/pull/17993

With almost 90% probability, there is no bug. But sometimes when the battery is connected, DSHOT does not initialize, there is no telemetry with ESC, the temperature is not displayed on the OSD, and RPM and Amp are not written to the logs.
And I assembled the second plane, where there is a regular GPS module. Mfv this bug appears periodically.

andyp1per, 2021-08-09:
This is actually a reporting issue. The problem is that CAN is very slow to start and means that BLHeli (which gets initialized late) has not been initialized by the time the banner is printed. I have fixed this in https://github.com/ArduPilot/ardupilot/pull/18266

closed 2021-08-14 by andyp1per