ArduPilot/ardupilot #17702: Copter: PSC.VX "vibration"

rmackay9, 2021-06-08:
Copter-4.1 with SCurves is showing a sort of "vibration" in the position controller target position.
[image]

This can be reproduced in SITL quite easily by creating a mission with a long straight segment.
[image]

priseborough, 2021-06-09:
I don't believe this is an EKF issue. If we log the PSC data at 400Hz and inspect the position as received by the controller, no noise is seen in the position, eg:

Here's the 'buzz' in demanded velocity:

[image]

Here's the target and measured position:

[image]

rmackay9, 2021-06-09:
Ok, thanks.  It seems likely that the little jumps I show in the position graph above are because I'm graphing line number vs position instead of time vs position.

lthall, 2021-06-09:
[image]

You can see when I differentiate the aircraft position the rate of change of that position varies a lot from sample to sample. I suspect this has something to do with the ekf using different numbers of samples and therefore different delta times while the position controller is assuming a uniform time....

lthall, 2021-06-09:
Close up:
[image]

lthall, 2021-06-09:
Could this be something like the time the ekf is calculating the position at is not the time of the next expected 400 Hz loop?

As the EKF update time and 400 Hz time moved together and apart it may look something like this....

priseborough, 2021-06-10:
@lthall The EKF inertial nav in the output predictor is using a downsampled data product that on each autopilot atomic time step is using a different number of contributing IMU samples. For example if down sampling from 1000Hz sensor data to 400Hz autopilot rate, the number of samples will alternate between 2 and 3. As such the EKF output is correct at the time stamp of the IMU data it receives. Possible solutions include:

1) Change the autopilot or IMU sample rate so that the IUMU sample rate is a integer multiple of the autopilot
2) Time stamp the EKF solution passed via the ahrs or inav interface using the timestamp of the latest IMU sample that was used by the EKF. The position controller would then need to account for the irregular time step.

lthall, 2021-06-11:
@priseborough Yeh I suspect the best approach is the translate the kinematics from the last ekf timestamp to the expected timestamp of the current fast loop. Given this results in a small 200 Hz noise signal the filters do a good job of removing it. So it isn't something that will crash aircraft.

peterbarker, 2021-06-11:
I'm a bit confused as to why the 10Hz log doesn't have these artifacts.  The 10Hz log should simply be a subsampled version of the 400Hz log AFAIK - and yet it doesn't have the same furry spots - 'though it still has the associated discontinuities.

lthall, 2021-06-12:
A 10 Hz log will have the same artifacts but they will be more difficult to see because you only have a 50% chance of getting the offset value when the fuz is present and then you only get 1 every 40 samples so the change in line slope almost disappears.

Talking through this with Randy I think the best approach moving forward is to move the time horizon of the position and navigation controllers back to the last EKF update time. This will then remove the problem of both timing jitter in the EKF update and the scheduling of the 400 Hz fast loop.

Filters will then be run with a timing jitter but this can be treated as noise.

This would mean that we must pass the time since the last EKF update acted on during the fast loop, into all objects that integrate or differentiate anything. So most PID objects and all attitude, position and navigation objects.

lthall, 2021-07-30:
I wonder if this will be solved with the increase in precision of the position variables.

rmackay9, 2021-07-30:
@lthall,

The vibration still appears I'm afraid.
[image]

By the way, for others watching, this reproduction of this issue requires moving these three lines found in Copter.cpp from the "ten_hz_logging_loop()" to the "fourhundred_hz-logging()" method.

```
    if (should_log(MASK_LOG_NTUN) && (flightmode->requires_GPS() || landing_with_GPS() || !flightmode->has_manual_throttle())) {
        pos_control->write_log();
    }
```

lthall, 2021-08-03:
I think we can close this. This is a 1cm/s noise at a very high frequency (something like 200 Hz). The E filter in the velocity controller will all but remove this.

closed 2021-08-03 by rmackay9