ArduPilot/ardupilot #17706: Copter: Copter does not slow down vertically while precision landing is active if downward facing rangefinder is not present

rishabsingh3003, 2021-06-08:
## Bug report

**Issue details**
Copter slows down vertically when it approaches the landing target if a downward-facing rangefinder is configured. However, a rangefinder is only compulsory while using IR-Lock landing system. If we have a companion computer (mavlink based) precision landing, which specifies the "distance" field, we do not need a rangefinder for landing. 

Therefore the speed will never reduce for a companion computer-based precision landing and the end result is that the copter may reach the ground before it reaches the landing target horizontally.

**Version**
Master ( and probably since many years) 

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rishabsingh3003, 2021-06-08:
Yes, you are probably right. That's why no one noticed it till now!
I'll fix it soon, should be a easy fix

amilcarlucas, 2021-06-08:
Strange... we use a companion computer for vision based precision landing and it emits MavLink DISTANCE_SENSOR messages, and it does reduce the lading speed near the ground ... 
Or am I imagining things ??

rishabsingh3003, 2021-06-08:
@amilcarlucas if you emit a DISTANCE _SENSOR that has downward facing orientation, then you're basically just mocking a rangefinder (I.e rangefinder lib reads this as a valid rangefinder driver). You'll get all functionality you'll normally get from any other downward facing rangefinder.