ArduPilot/ardupilot #17707: VTOL tailsitter turning towards first wapoint during conversion in auto mission.

dcamatti, 2021-06-08:
## Bug report

**Issue details**

During automatic mission tested on our VTOL vectored tailsitter we discovered that vehicle starts navigating towards first mission waypoint as soon as it changes for multicopter to FW. In other words, when the transition angle is achieved and attitude measurement s switched from MC to FW, desired roll jumps to the value required to make the plane turn onto first waypoint. This force the vehicle to turn before completing the transition actually causing a great destabilization of flight.

**Version**
ArduPlane V4.1.0dev (74045ba5)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Two motors, vectored tailsitter

**Hardware [type**](url)
Cube orange

**Logs**
[2021-06-08 11-23-57.zip]
(https://github.com/ArduPilot/ardupilot/files/6619456/2021-06-08.11-23-57.zip)

robustini, 2021-06-08:
@dcamatti it is a known problem, never fixed.
I hope it's the right time to fix it! 👍

IamPete1, 2021-06-08:
This is the transition in question, as soon as the transition is complete the desired roll jumps up to 25 deg. This is expected bahavour not a bug, but certainly something we could improve.

[image]

There is also a issue in back transition, this one can be fixed with params. Your transition is timing out, this is because we never reach the target transition angle. 

[image]

There are a few fixes, best is to improve the forward flight tune so that the pitch better tracks the target. Your carrying a large nose down I term for the fixed wing flight, so CG / servo trim should be able to help remove this. If with a better tune you still cant keep up you could reduce the transition pitch rate, or lower the target angle.

dcamatti, 2021-06-09:
Dear Peter,

> 
> This is the transition in question, as soon as the transition is complete the desired roll jumps up to 25 deg. This is expected bahavour not a bug, but certainly something we could improve.

I understand your point of view which is related to the pure coding considerations, but for me, everything that is preventing a safe flight is a bug. I saw that the "transition done" state is flagged exactely when the 90 deg attitude offset is applied, i.e. the transition angle is reached. If the vehicle reaches the transition angle we cannot be sure that the transition is done. There still are a certain number of problem that may arise. More, in this way we are issuing commands (turns) when the vehicle is doing something else (completeting transition from transition angle to level flight) which still is a critical phase like a post takeoff condition in a normal plane. We cannot disturb the vehicle during this phase. I think that transition has to be considered completed at least after Airspeed > min Cruise speed plus a certain time delay. Several other checks could be done in order to validate the "transition done" estimation. 
- 
> 
> [image]
> 
> There is also a issue in back transition, this one can be fixed with params. Your transition is timing out, this is because we never reach the target transition angle.

We are aware of this. Transition angle for this flight was set at 75 deg which were not completely reached. Can you explain which is the condition who is deciding to proceed with the conversion even is the angle is not reached yet? In there a variable controlling this?
> [image]
>
> There are a few fixes, best is to improve the forward flight tune so that the pitch better tracks the target. Your carrying a large nose down I term for the fixed wing flight, so CG / servo [...]

Thanks for this suggestions. We already made several tunign flight to improve pitch loop which is performing quite well in cruise. We are investigting right now if we are hitting some I antiwindup somewhere. We cannot move neutral controls point away of zero angle condition otherwise we encounter problem in copter mode. (q trim is not working, there is another issue for that...).
Regarding the transition angle we need to keep it very high (>70 deg) otyherwise the motors suddently tilt up causing a big stess in the structure.

Many thanks in advance.

Daniele

dcamatti, 2021-06-09:
Sorry bold font, cannot take rid of that....not wanted

IamPete1, 2021-06-09:
re-bug vs not bug. A bug is where the code does something we did not mean it to. In this case the code is doing what we expect , yes there is lots of scope for improving, but it is not going wrong.

> Can you explain which is the condition who is deciding to proceed with the conversion even is the angle is not reached yet? In there a variable controlling this?

The timeout is set based on a 1.5 times the expected time transition. This expected time is calculated from the initial pitch angle, the transition pitch angle and the transition rate.

dcamatti, 2021-06-09:
OK thanks, bug Vs not bug it is a useless debate. I understand the code is doing as expected; the problem is that doing this way is highly probable we will crash the vehicle if we not take care of alighing it toward the first waypoint. Nevertheless it is not good to see at all. 
What it is important is to understand a viable solution in order to prevent the vehicle start navigating as soon as the pitch 90 deg shift is applied. 
Question: is the navigation unlocked by the "transition done" flag?
Some ideas (dont know which one is really viable..):
1 - use a timeout system (similar to back conversion) in order to delay the "transition done" flag.
2 - introduce a "temporary cruise phase", after the pitch shift is applied, in order to have the vehicle navigate at constant altitude and heading for a while, then resume normal navigation. This phase shuld not be too long in order to prevent vehicle moving at transition altitude too much, but long enought in order to let vehicle stabilize and completely terminate all transient events from recent transition. I think that something around 2 seconds could be more than enought.
3 - (similar to 2): force pitch at fixed value and heading at the value assumed during transition for a fixed amount of time and then resume normal navigation.

Regarding back transition, I would like to thank you, now everything is clear. We discovered that our vehicle of not capable of exibithing more than 40/45 deg/sec of pitch rate. So we need to flatten the pitchup ramp in order to prevend accumulating too much delay and triggering the timeout.

Many Thanks.

Daniele

closed 2022-01-26 by tridge