ArduPilot/ardupilot #17709: _brake_accel Not Reset

ccackam, 2021-06-08:
## Bug report

**Issue details**

While flying Arducopter recently we swapped from RTL to Loiter and then to auto. We only spent about 3 seconds in loiter so the decelerations didn't damp out until after the switch to auto. Later in the flight we again swapped to loiter from RTL and again had very large decelerations. However, this time they were much larger. We've flown this system for 2 years with no issues. So I think something about this particular situation caused the larger deceleration. In both cases the pilot didn't deviate the sticks. I think our investigation may have revealed a minor bug. 

In the file [AC_Loiter.cpp](https://github.com/ArduPilot/ardupilot/blob/058d4299ade75677727d2cf4cdc7b1f6cb8b8b5a/libraries/AC_WPNav/AC_Loiter.cpp#L258) and associated header file [AC_Loiter.h](https://github.com/ArduPilot/ardupilot/blob/058d4299ade75677727d2cf4cdc7b1f6cb8b8b5a/libraries/AC_WPNav/AC_Loiter.h#L89) the member variable `_brake_accel` is never reset to 0. So our working theory is that the residual value of `_break_accel` which was left over from the previous 3 seconds in loiter mode increased the magnitude of the deceleration. The crux of the issue is this line here:

```cpp
_brake_accel += constrain_float(loiter_brake_accel-_brake_accel, -_brake_jerk_max_cmsss*nav_dt, _brake_jerk_max_cmsss*nav_dt);
```

While this code does saturate the change in `_brake_accel`, it does not care what _brake_accel is originally. We are also not sure where the value _brake_accel is initialized to a starting value, but we assume that the object type AP_Float takes care of this. 

A potential quick fix for this would be to reset `_brake_accel` to 0 in the `init_target` method of the AC_Loiter class, but we wanted to consult with people who know the code better before making the change.

**Version**
4.0.7

**Platform**
[x] All
[  ] AntennaTracker
[Observed] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X Config Hexacopter

**Hardware type**
H7 Blue Cube 

**Logs**
N/A

rmackay9, 2021-06-09:
This is a really great find, thanks!  I've created PR https://github.com/ArduPilot/ardupilot/pull/17717 with a potential fix.  If all goes well this will be included in Copter-4.1.0-beta4. I think it's unlikely we will back port it to 4.0.

ccackam, 2021-06-09:
Glad we could help

mschnur, 2021-06-21:
@rmackay9 I work with @ccackam and was reviewing #17717, and one thing that caught my eye was that `_brake_timer` isn't being initialized when you initialize `_brake_accel`, nor is it set anywhere else other than in this block of code:
https://github.com/rmackay9/rmackay9-ardupilot/blob/171f13fbdf586d2df2b579bb63c855a3936822c3/libraries/AC_WPNav/AC_Loiter.cpp#L249-L259

I'm not particularly familiar with the inner workings of ardupilot, but from the block of code linked above it looks to me like it should be possible to read and use an uninitialized value from `_brake_timer`. For instance, if the pilot takes off without ever entering loiter mode, then enters loiter mode with the sticks at zero, then I believe `_brake_timer` would be uninitialized when it is used in
```
if ((AP_HAL::millis()-_brake_timer) > _brake_delay * 1000.0f)
```
I don't believe `_brake_timer` could cause _anywhere_ near as severe an issue as we saw with the uninitialized `_brake_accel`, but I do think it could cause the autopilot to disobey the `_brake_delay` parameter value. If the garbage value read from `_brake_timer` is greater than the return value from `AP_HAL::millis()`, then the subtraction will underflow since they're now both unsigned integers and the result will be a presumably very large integer. This means the code would likely immediately enter the inner `if`. Note that previously, when `_brake_timer` was a float, the return value of `AP_HAL::millis()` would have been implicitly promoted to a float prior to the subtraction, and the result would be a negative floating point number so the inner `if` wouldn't be entered. If the garbage value read from `_brake_timer` is significantly less than the return value from `AP_HAL::millis()` then the code would likely immediately enter the inner `if`.

I'm sure you'll have a much deeper understanding of the ramifications of this behavior, but at a glance as someone unfamiliar with it what I described above seems like anomalous behavior.

rmackay9, 2021-06-22:
@mschnur, thanks very much for looking into the details of this.

I can't answer all your questions but _brake_timer is initialised to zero automatically.  I think the way it works is AP initialises variables created on the stack with zero values while local variables are not initialised.  This means that class members in general don't need to be initialised to zero.  The reason we do this is that we are always trying to keep AP's code size as small as possible so that we can fit more features on some boards with flash limitations.

Knowing this do you think the logic is OK?

mschnur, 2021-06-22:
@rmackay9 I was basing my analysis of this on the C++ standard. Per that standard, `_brake_timer` is default initialized because it is a "non-static data member [that] is not mentioned in a constructor initializer list and that constructor is called." Since it is not a class or an array, that means "nothing is done: the objects with automatic storage duration (and their subobjects) are initialized to indeterminate values." [Source](https://en.cppreference.com/w/cpp/language/default_initialization).

If AP behaves in a way different from the standard as you describe, then yeah I think it's fine as-is.

mschnur, 2021-06-22:
@rmackay9 Sorry to keep pinging you about this, but this should be my last question. I'm trying to understand the non-standard zero-initialization you described. Specifically, how it happens. I found [`libraries/AP_Common/c++.cpp`](https://github.com/ArduPilot/ardupilot/blob/d82a229584d9872c84c06995497759904b077fa5/libraries/AP_Common/c++.cpp), and I understand how that global override of `new` will cause dynamically allocated objects to be zero-initialized (which in turns means their members will be zero-initialized). Specific to this case (i.e. `_brake_timer`), I also see that [the `AC_Loiter` object used globally in ArduCopter is dynamically allocated](https://github.com/ArduPilot/ardupilot/blob/d82a229584d9872c84c06995497759904b077fa5/ArduCopter/system.cpp#L546), meaning it will be zero-initialized as you described.

Is the mechanism for the zero-initialization you described previously? If so, am I correct in understanding that the zero-initialization only applies to dynamically allocated objects (and dynamically allocated memory in general), and that statically allocated objects (and statically allocated memory in general) will still be initialized according to the C++ spec? In other words, I haven't found any mechanism that would zero-initialize statically-allocated memory -- am I just missing it or does it not exist?

rmackay9, 2021-06-22:
@mschnur,

No problems with more questions!   The statically allocated variables and objects are also initialised to zero.  I don't delve into these depths of AP but I'm sure it's happening.

PR https://github.com/ArduPilot/ardupilot/pull/17800 is an example where we've removed initialisation to save some code space because the init isn't necessary.

IamPete1, 2021-08-08:
@rmackay9 your fix for this has been merged, so I think this can be closed?

rmackay9, 2021-08-08:
Yes, fixed, thanks!

closed 2021-08-08 by rmackay9