ArduPilot/ardupilot #17714: About Copter ekf_check() -> ekf_over_threshold

LeiOuYang, 2021-06-09:
version: Copter-4.0.7

vel_variance, position_variance, and  mag_variance were checked, but height_variance were not checked!

I recently encountered altitude anomaly in my flight, so I paid attention to this code.

I want to know the reason why the height check is not included in the EKF check.

Thanks!

rmackay9, 2021-06-09:
@LeiOuYang,

This is a good question .. but the simple answer could be, "what would we do about it?".  If the ekf check fails, the EKF failsafe action is to switch the vehicle to either Land (the default) or AltHold but both of these modes require a good altitude and climb rate estimate.  Even if the EKF's height estimate goes bad, the EKF failsafe will likely not help.

Instead what we have is the [Vibration Failsafe](https://ardupilot.org/copter/docs/vibration-failsafe.html) that tries to regain a good height estimate (and control) by changing the method that the EKF uses to do the estimation.

Great question but it's also a support question so I will close this issue if that's OK. I will still monitor this issue for replies though.

closed 2021-06-09 by rmackay9

LeiOuYang, 2021-06-09:
@rmackay9 
Thank you for your reply. 
I thought there was some reason why I didn't check the height. I know now. 
You can close the issue, thank you！