ArduPilot/ardupilot #17724: ArduSub not disarming when communication lost

ES-Alexander, 2021-06-10:
## Bug report

**Issue details**

Vehicle not disarming once communication lost/tether unplugged - can cause thrusters to continue trying to stabilise while operator is retrieving ROV from the water. Discussion on Blue Robotics forum [here](https://discuss.bluerobotics.com/t/disarm-or-manual-mode-when-communication-is-lost/7804/6)

**Version**
_What version was the issue encountered with_
bluerobotics companion 0.0.26 w/
- ArduSub 4.0.3
- ArduSub 4.0.2

bluerobotics companion 0.0.22 w/
- (at least) ArduSub 4.0.3

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[x] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Assuming standard BlueROV2 - vectored

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Pixhawk 1

**Logs**
_Please provide a link to any relevant logs that show the issue_
Dataflash logs provided by user - includes one log per companion/ArduSub combination, and a vehicle params file to confirm Pilot Input and GCS Heartbeat failsafes are enabled and set to 'Disarm'.
[OneDrive-2021-06-09.zip](https://github.com/ArduPilot/ardupilot/files/6628270/OneDrive-2021-06-09.zip)

peterbarker, 2021-06-11:
```
param show FS_GCS_*
MANUAL> FS_GCS_ENABLE    2.000000

MANUAL> 
MANUAL> arm throttle
MANUAL> Got COMMAND_ACK: COMPONENT_ARM_DISARM: ACCEPTED
AP: Arming motors
ARMED
Arming checks disabled

MANUAL> 
MANUAL> set heartbeat
heartbeat
MANUAL> set heartbeat 0
MANUAL> AP: MYGCS: 255, heartbeat lost
AP: Disarming motors
DISARMED
```

This is on master.

So the problem isn't trivially reproducible.

peterbarker, 2021-06-11:
.... same happens (vehicle disarms) in SITL in the git hash present in the log files.

I'm wondering if the vehicle might be receiving heartbeats or manual-control or whatnot packets from somewhere else.

A reproduction in SITL would be convenient for tracking this down.

jaxxzer, 2021-06-14:
@peterbarker it does appear to be some configuration specific issue, and the user might have some mavlink messages on board that we were not expecting. I agree that on the bench, the failsafe appears to be working as designed, thanks for verifying that!

patrickelectric, 2021-07-15:
The problem was a mavlink based service running on the companion computer with GCS system_id.

closed 2021-07-15 by patrickelectric