ArduPilot/ardupilot #17733: AP_GPS : add parameter to still be able to inject RTCM data to a GPS configured as GPS_TYPE,18 (GPS_TYPE_UBLOX_RTK_ROVER)

lebarsfa, 2021-06-10:
## Feature request

**Is your feature request related to a problem? Please describe.**
The line https://github.com/ArduPilot/ardupilot/blob/0b517cfc57330edea8ed694472c4b0d30d7f49b3/libraries/AP_GPS/AP_GPS.cpp#L1219 is preventing RTCM data to be sent to a GPS configured as a rover in a moving baseline scheme, which makes sense. However, in case we have an already configured rover and moving base which already communicate together (see e.g. https://www.ardusimple.com/ardupilot-simplertk2bheading-configuration/) and we would like the moving base to get RTCM data from Mission Planner through telemetry, it could be useful to be able to use a single serial port on the Pixhawk (especially the Pixhawk 4 mini does not have many serial ports) where the TX would be connected to the moving base (to send RTCM) and the RX would be connected to the rover (to get position, yaw, etc.).

**Describe the solution you'd like**
Is it possible to add a parameter that disables the lines https://github.com/ArduPilot/ardupilot/blob/0b517cfc57330edea8ed694472c4b0d30d7f49b3/libraries/AP_GPS/AP_GPS.cpp#L1219-L1222
It could be just adding a special value for GPS_INJECT_TO parameter, e.g. 255.

**Describe alternatives you've considered**
It seems possible to temporarily change GPS_TYPE to 1 (instead of 18) without rebooting the Pixhawk to get the intended behavior, but this might have side effects.
Otherwise, it should be possible to use 2 serial ports on the Pixhawk, one dedicated to the rover and the other to the moving base. The drawback is that we use 2 serial ports while only one could be used.

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2022-09-10:
A rover can only get corrections from a single base station. When being used for yaw this is the other GPS on the vehicle. That base GPS can receive corrections from another base, via mission planner.

lebarsfa, 2022-09-10:
I confirm that I did not ask to send corrections from multiple base stations to a single rover. If I remember well, what I proposed was to be able to use a single serial port for 2 different purposes : 

1. Sending corrections (RTCM coming from Mission Planner) to the moving base.
2. Getting the yaw and position from the rover. Note that in the use case I describe, the moving base is already connected (to send RTCM for yaw computation) to the rover outside of the Pixhawk, thanks to the ArduSimple board (but it can also be done easily without that specific board, it is just a matter of connecting a TX to a RX).

At the time of opening the issue (I don't know if there were changes since then), 2 serial ports on the Pixhawk were necessary, while it should be possible to use only 1  with a minor code change.

[image]