ArduPilot/ardupilot #17747: low LOIT_ACC_MAX leads to incorrect loiter speed

tridge, 2021-06-12:
This was first noticed in quadplanes, where low values of Q_LOIT_SPEED and Q_LOIT_ACC_MAX are common, but also happens in copter and may especially impact large vehicles.
When LOIT_ACC_MAX is low (say a value of 50) then the speed constraint in LOIT_SPEED is ineffective. For example, with LOIT_SPEED=300 and LOIT_ACC_MAX=50 the default SITL copter will reach over 6m/s in LOITER mode. 
This is a regression from 4.0
[image]
the above graph shows the default copter model with SITL. It has LOIT_SPEED=300, and the graph shows the impact of changing LOIT_ACC_MAX from a start value of 50 in steps of 50 up to 250

lthall, 2021-06-12:
The problem here is that an extremely low value of maximum acceleration correction was used in the plane parameters. The default is 
LOIT_ACC_MAX,500 or approximately 27 degrees.

This means the position controller can lean up to 27 degrees from the commanded attitude to make the aircraft follow the loiter drag model or fight against wind.

Plane was using 50, or approximately only 2.9 degrees. This meant that the commanded lean angle could not be reduced to slow the aircraft down.

So the basic problem is the parameter was not set to a sensible number and changed from my suggested default.

Now this parameter was not implemented correctly initially and did not do what it was supposed to do. This is why it didn't cause a problem initially.

I will need to think on this further.......

lthall, 2021-06-12:
After thinking about this I think the best approach is to stay consistent with the previous behaviour and have this variable only impact the position controller approach acceleration limit.

I will fix this problem....

lthall, 2021-06-20:
This should fix this problem:
https://github.com/ArduPilot/ardupilot/pull/17815

PR #17815 (2021-06-21): AC_PosControl: remove pos control accel correction limit
This PR fixes #17747

This is a minor change that only affects loiter. This change restores the previous behaviour of Loiter and removes the unnecessary additional variable.
closed 2021-06-21 by rmackay9