ArduPilot/ardupilot #17763: PreArm: Internal errors 0x100000 l:868 flow_of_ctr

peterbarker, 2021-06-14:
## Bug report

**Issue details**

During flight a H743-based board received this internal error.

That's this internal error:
```
void AC_PosControl::update_z_controller()
{
    // Check for z_controller time out
    if (!is_active_z()) {
        init_z_controller();
        if (has_good_timing()) {
            // call internal error because initialisation has not been done
            INTERNAL_ERROR(AP_InternalError::error_t::flow_of_control);
        }
    }
    _last_update_z_us = AP_HAL::micros64();
```

**Version**

67fea9f2

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad X

**Hardware type**

H743-Slim

**Logs**

On demand

rmackay9, 2021-06-14:
Thanks for the report.  I think it would be better if we created a new internal error for this failure to reduce the support load.  It really isn't a flow-of-control problem, it's a loop timing problem.

MichelleRos, 2021-06-14:
Here's the log for that flight: https://drive.google.com/file/d/1NH23R-71nb25ORF26DxB3lrTnsyodWsu/view?usp=sharing

Williangalvani, 2021-06-22:
I'm having the same issue running Sub on a Navigator (our linux board).
Commit d82a22958.

I can engage althold once. The second time it gives me this internal error.

rmackay9, 2021-06-23:
@Williangalvani,

This internal error catches the situation in which the position controller is run after a break of at least 12ms (or 5 main loop updates) without having first been initialised.  The two likely causes are:

1. there is a software bug in the althold mode where it is not initialising the position controller
2. the main loop is getting stuck for more than 12ms sometimes (maybe check the PM message)

It is very possible that this check if too strict but it might be good to investigate the above before we relax it.

spiderkeys, 2021-06-29:
Also seeing this issue in Sub (same symptoms as @Williangalvani) running on the MatekH743-Wing. I am deep in the weeds instrumenting the loop performance, so I will try to surface some data on whether or not that is the culprit.

spiderkeys, 2021-07-03:
From what I have observed so far, it seems like this ended up being a performance/timing issue, as posited by @rmackay9 . I had significant delays throughout the scheduler due to some bugs around Dshot/digital motor timing, and once I resolved those and got my fast loop and control loops back to normal speed, this doesn't seem to happen anymore. YMMV.

spiderkeys, 2021-07-06:
Update: The issue is still there for me in sub, I had just had logging turned off in my last tests, and dataflash log setup upon arming is a consistent source of 18-30ms delays that violate the timing thresholds in the AC_PosController timeouts, at least on Sub which currently has its fast_loop rate set to 400Hz. At 400Hz, the max allowed delta for poscontroller updates is currently 12.5ms (5 * 1/400hz).

I have temporarily increased the fast loop time scalar to 50x, rather than 5x to alleviate this from happening in my current setup, though this is just a workaround.

Williangalvani, 2021-08-05:
My issues were really just timing issues. switching kernels and tweaking i2c speeds I managed to get rid of the issue.

IamPete1, 2021-09-12:
According to the copter issues list this has been fixed by https://github.com/ArduPilot/ardupilot/pull/18416

closed 2021-09-12 by IamPete1