ArduPilot/ardupilot #17767: 4.1b3 AP_Logger stuck thread (fsync)

davidbrodrick, 2021-06-14:
## Bug report

**Issue details**

Message is reported on the GCS "AT_Logger: stuck thread (fsync)" and then any further logging to the log file ceases. This has happened twice in the last few flights.

**Version**

4.1beta3

**Platform**
[  ] All
[  ] AntennaTracker
[ X  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

Quad X

**Hardware type**

Pixhawk

**Logs**

Logs stop once the message is reported.

peterbarker, 2021-06-17:
@davidbrodrick this may be SD card corruption - the filesystem layer just not coping with something going wrong.

Please try another SD card and see if things continue to go wrong.

Which board is this?

rmackay9, 2021-07-12:
We think this has been fixed in Copter-4.1.0-beta5 so if you could re-test with that it would be great.  I'm going to close this issue now but I'll monitor it for responses so we can re-open if necessary.

closed 2021-07-12 by rmackay9